project-chip/connectedhomeip #19518: [Pre-SVE] TC-SU 2.2 step 4 -t does not make DUT wait when updateAvailable

HubertHugo, 2022-06-13:
platform: nRF52840
all-clusters-app
SHA: 5d8599d195a93dfe5473f64e4f888c322e5df1d2
In test 2.2 step 4 running OTA with command:
```
./chip-ota-provider-app --filepath ~/connectedhomeip/examples/all-clusters-app/nrfconnect/build4/zephyr/matter.ota -q updateAvailable -t 180
```
and announcing does not make DUT wait for update. It starts updating right after announcing.

carol-apple, 2022-06-13:
There's been a recent spec change:
```
The DelayedActionTime field SHALL only be present if the Status field is set to Busy.
```

ukatira, 2022-06-13:
OTA Test plan was updated to account for recent spec changes - https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5248
And this test case has been removed as a part of this change.