project-chip/connectedhomeip #19521: Switch failed to connect with Light after only Light was rebooted

shubhamdp, 2022-06-13:
#### Problem
Environment:
* Commit : c1e1d018820c0fdd51f124f607b9207f31fae2ff
* Light : examples/lighting-app/esp32
* Switch : examples/light-switch-app/esp32

Steps to reproduce:
```
# Commission the Light
./chip-tool pairing ble-wifi 1 ES_Staff ES@puba45 20202021 3840

# Commission the Switch
./chip-tool pairing ble-wifi 2 ES_Staff ES@puba45 20202021 3840

# Install ACLs on Light
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null }, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": [2], "targets": [{"cluster": 6, "endpoint": 1, "deviceType": null}, {"cluster": 8, "endpoint": 1, "deviceType": null}]}]' 1 0

# Install Bindings on Switch
./chip-tool binding write binding '[{"fabricIndex": 1, "node": 1, "endpoint": 1, "cluster": 6}, {"fabricIndex": 1, "node": 1, "endpoint": 1, "cluster": 8}]' 2 1
```
* On Switch command line try `matter switch onoff on` and `matter switch onoff off` commands to control the Light.
* Reboot the Light and try the on/off commands again

Actual behavior
* Switch is unable to establish CASE session with Light
* Light 
```
E (95471) chip[IN]: Data received on an unknown session (LSID=16996). Dropping it!
E (101711) chip[IN]: Data received on an unknown session (LSID=16996). Dropping it!
E (107441) chip[IN]: Data received on an unknown session (LSID=16996). Dropping it!
E (116961) chip[IN]: Data received on an unknown session (LSID=16996). Dropping it!
E (131081) chip[IN]: Data received on an unknown session (LSID=16996). Dropping it!
```

* Switch
```
> matter switch onoff on
Done
I (340594) chip[IN]: Prepared secure message 0x3ffc792c to 0x0000000000000001 (1)  of type 0x8 and protocolId (0, 1) on exchange 16943i with MessageCounter:150399514.
I (340604) chip[IN]: Sending encrypted msg 0x3ffc792c with MessageCounter:150399514 to 0x0000000000000001 (1) at monotonic time: 0000000000052F97 msec
> I (346854) chip[IN]: Sending encrypted msg 0x3ffc792c with MessageCounter:150399514 to 0x0000000000000001 (1) at monotonic time: 00000000000547FE msec
I (347054) chip[DIS]: UDP:[FE80::7EDF:A1FF:FEA0:27F0%st1]:5540: new best score: 3
I (347054) chip[DIS]: UDP:192.168.20.34%st1:5540: score has not improved: 2
I (347064) chip[DIS]: Checking node lookup status after 195 ms
I (347064) chip[DIS]: Keeping DNSSD lookup active
I (347074) chip[DIS]: Discovered node without a pending query
I (347084) chip[DIS]: UDP:[FE80::7EDF:A1FF:FEA0:27F0%st1]:5540: score has not improved: 3
I (347094) chip[DIS]: UDP:192.168.20.34%st1:5540: score has not improved: 2
I (347094) chip[DIS]: Checking node lookup status after 223 ms
I (347104) chip[DIS]: Discovery does not require any more timeouts
I (352554) chip[IN]: Sending encrypted msg 0x3ffc792c with MessageCounter:150399514 to 0x0000000000000001 (1) at monotonic time: 0000000000055E41 msec
I (362174) chip[IN]: Sending encrypted msg 0x3ffc792c with MessageCounter:150399514 to 0x0000000000000001 (1) at monotonic time: 00000000000583D3 msec
I (367594) chip[DMG]: Time out! failed to receive invoke command response from Exchange: 16943i
E (367594) chip[-]: OnOff command failed: 32
I (376244) chip[IN]: Sending encrypted msg 0x3ffc792c with MessageCounter:150399514 to 0x0000000000000001 (1) at monotonic time: 000000000005BACF msec
E (399064) chip[EM]: Failed to Send CHIP MessageCounter:150399514 on exchange 16943i sendCount: 4 max retries: 4
I (434954) chip[DIS]: Discovered node without a pending query
I (434954) chip[DIS]: Discovery does not require any more timeouts
```

Logs
* [switch.log](https://github.com/project-chip/connectedhomeip/files/8890452/switch.log)
* [light_after_reboot.log](https://github.com/project-chip/connectedhomeip/files/8890453/light_after_reboot.log)
* [light_before_reboot.log](https://github.com/project-chip/connectedhomeip/files/8890454/light_before_reboot.log)
* [chip_tool.log](https://github.com/project-chip/connectedhomeip/files/8890455/chip_tool.log)

 #### Proposed Solution
<suggested fix, suggested enhancement>

shubhamdp, 2022-06-13:
@bzbarsky-apple @tcarmelveilleux is this expected? If yes, then how can we fix this?

bzbarsky-apple, 2022-06-13:
This is expected so far until #17568 and/or https://github.com/project-chip/connectedhomeip/issues/17298 are fixed.

dhrishi, 2022-07-21:
@mrjerryjohns @bzbarsky-apple Is this expected to work now? #17568 is fixed and a part of #17298 is open yet.

bzbarsky-apple, 2022-07-21:
@dhrishi On tip as of today the expectation is that the first communication attempt after the reboot fails, and subsequent ones work.

dhrishi, 2022-07-23:
@bzbarsky-apple So the subsequent communication/request needs to be sent out explicitly? Or it happens internally as a retry attempt?

bzbarsky-apple, 2022-07-23:
It needs to be sent out explicitly.

dhrishi, 2022-07-25:
Ok. Thanks!

closed 2023-02-04 by stale[bot]