project-chip/connectedhomeip #19535: `subscriptionsPerFabric` in basic cluster has incorrect values

mrjerryjohns, 2022-06-13:
#### Problem

Reading out `subscriptionsPerFabric` gives a really large number (which might be accurate on heap-based platforms anyways), but looked suspicious.

Looks like the logic is off in `InteractionModelEngine::GetMinSubscriptionsPerFabric()`:

```
    size_t perFabricSubscriptionCapacity = static_cast<size_t>(readHandlerPoolCapacity - kReservedPathsForReads) / fabricCount;
```

That should be `kReservedHandlersForRead`, not paths.

Also, `GetReadHandlerPoolCapacity` is not quite handling heap-based platforms today, resulting in an advertised number lower than the capability of the platform.

mrjerryjohns, 2022-06-21:
Last of all, the computation of the value is dividing the total available resources by the number of currently commissioned fabrics. This is not compliant with what the spec is actually saying here (which is requiring the minimum number of subs supported per fabric assuming at least 5 fabrics could be commissioned on the device).

bzbarsky-apple, 2022-06-21:
So we need two functions: the bare minimum we guarantee per fabric (which depends on max fabrics), and the value the resource-allocation logic needs, which is depends on our actual number of fabrics.

erjiaqing, 2022-06-23:
Note: Since #18884 touched the macros for resource minimals, the original issue is fixed already.

PR #20207 (2022-07-03): [IM] Update GetMinSubscriptionsPerFabric
#### Problem

Fixes #19535 

#### Change overview

- Renamed `GetMinSubscriptionsPerFabric` to `GetMinGuaranteedSubscriptionsPerFabric`
- The updated code is a solution that will never be wrong. (Return a constant value)
- Updated     static constexpr size_t kMinSupportedSubscriptionsPerFabric = 3; static constexpr size_t kMinSupportedPathsPerSubscription   = 3; Which should be updated in previous PRs, but I think it is also suitable to include the change in this PR

#### Testing

- Build with `#define CHIP_SYSTEM_CONFIG_POOL_USE_HEAP 1` it returns 6553 [... cut]