project-chip/connectedhomeip #19558: [TC-MC-6.4]- DUT is not returning SPEED_OUT_OF_RANGE (status 4) error code when TH sends consecutive FastForward and Rewind commands

Rajashreekalmane, 2022-06-14:
#### Problem
As per Test-Plan test step 15 and 16 if TH sends Fast-forward and rewind command from to DUT , DUT should respond PlaybackResponse status has  SPEED_OUT_OF_RANGE (status 4) error code, But currently DUT responds PlaybackResponse status has 0.

 * expected behavior:-DUT should respond PlaybackResponse status has  SPEED_OUT_OF_RANGE (status 4) error code 
 * actual behavior:-DUT responds has 0
 
 * steps to reproduce:-
1.   ./chip-tool mediaplayback read  current-state  1 3
2. ./chip-tool mediaplayback read playback-speed 1 3
3. ./chip-tool mediaplayback fast-forward 1 3
4. ./chip-tool mediaplayback read  current-state  1 3
5. ./chip-tool mediaplayback read playback-speed 1 3
6. ./chip-tool mediaplayback fast-forward 1 3
7. ./chip-tool mediaplayback read playback-speed 1 3
8. ./chip-tool mediaplayback rewind 1 3
9. ./chip-tool mediaplayback read  current-state  1 3
10. ./chip-tool mediaplayback read playback-speed 1 3
11. ./chip-tool mediaplayback rewind 1 3
12. ./chip-tool mediaplayback read playback-speed 1 3
13. ./chip-tool mediaplayback  play 1 3
14.  ./chip-tool mediaplayback read playback-speed 1 3
15. ./chip-tool mediaplayback fast-forward 1 3
16. ./chip-tool mediaplayback rewind 1 3

 * system configuration
 system configuration
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP 
Commit-ID: c1e1d018820c0fdd51f124f607b9207f31fae2ff			
Dut application: tv-app
TH application:-Chip-tool
Platform:-Linux platform
 
Test-plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#tc-mc-6-4-variable-speed-media-playback-verification

Spec Reference:-
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/media/MediaPlayback.adoc#statusenum

PFA log below:-
[TC-MC-6.4 .txt](https://github.com/project-chip/connectedhomeip/files/8899424/TC-MC-6.4.txt)

bzbarsky-apple, 2022-06-14:
> expected behavior:-DUT should respond PlaybackResponse status has SPEED_OUT_OF_RANGE (status 4) error code

Why?  The test plan says:

> TH sends consecutive FastForward commands **up to the manufacturer defined max speed**

But the steps above don't do that.

chrisdecenzo, 2022-06-14:
re-opening because the example tv-app does not impose a max speed

closed 2022-06-15 by chrisdecenzo