project-chip/connectedhomeip #19561: [TC-PRS-2.3] getting invalid operands Error while checking MeasuredValue & ScaledValue attribute Range

kowsisoundhar12, 2022-06-14:
As per Testplan need to check MeasuredValue attribute range,In order to check this attribute we need following step 
     ----> (MeasuredValue – Tolerance) to (MeasuredValue + Tolerance)
     As per spec MeasuredValue data type is int16, Tolerance attribute data type is uint16 so getting operand type error because of this 2 different data type 

Notes:
SHA: 55ab764beabaed8675bd60e6d18e84d545a733e0
client app:  All-cluster-app
[PRS-2-3.zip](https://github.com/project-chip/connectedhomeip/files/8900236/PRS-2-3.zip)

server app: chip tool

Testplan Screenshot :
[image]

Spec Screenshot:
[image]

Error Screenshot:
[image]

bzbarsky-apple, 2022-06-14:
@kowsisoundhar12 Please attach the actual diff that was made to the YAML to get the above error message or link to that diff or PR on github.  That's the most useful part for being able to quickly reproduce this and test a fix (much more useful than screenshots of the spec that have nothing to really do with the actual problem, screenshots of errors that would be reproducible with the diff, or logs from a command that would get run anyway), and it's the one thing that is missing from this report...

bzbarsky-apple, 2022-06-14:
@kowsisoundhar12 And in general, for issue reports, the useful format is:
1. Steps X were performed.
2. The result was Y.
3. The expected result was Z.

This issue has item 2, and an implied item 3 ("should have no errors"), but is missing the key part of item 1, which is what changes were made to the tree before running regen and compiling.

kowsisoundhar12, 2022-06-14:
@bzbarsky-apple please find attached PRS 2.3.yaml file and log file here
[Test_TC_PRS_2_3.zip](https://github.com/project-chip/connectedhomeip/files/8900433/Test_TC_PRS_2_3.zip)
[PRS-2-3-Execution-logs.zip](https://github.com/project-chip/connectedhomeip/files/8900435/PRS-2-3-Execution-logs.zip)

1.I have used minValue and MaxValue constraint check for validating MeasuredValue & ScaledValue attribute Range

bzbarsky-apple, 2022-06-14:
@kowsisoundhar12 Thank you, that at least gives a way to reproduce. But again, a diff or link to diff would be a _lot_ easier to work with than a zipped-up entire file that then has to be unzipped and placed in the right place in the tree...

kowsisoundhar12, 2022-06-16:
@bzbarsky-apple  sorry about that, it does look like the verification step defined in  the test plan does not really check anything, let me check with the tests plan team on what was the reason behind this , I also noticed that these attributes tolerance ,Scale,ScaledTolerance, ScaledValue are giving unsupported attribute response in YAML framework and it seem to work during my testing due to the PICS conditions i had added, Ill look into this and file a follow up issue.

ReneJosefsen, 2022-06-19:
Yes, looking at the step is does not really make sense. I need to check with the original writer, but we will have to get the test plan updated.

kowsisoundhar12, 2022-07-12:
PRS-2.3 Testscript is removed from Testplan ,hence closing this issue

closed 2022-07-12 by kowsisoundhar12