project-chip/connectedhomeip #19563: Door-lock   Set User Command is returning success instead of  INVALID_COMMAND

Ashwinigrl, 2022-06-14:
**Problem** -In Set User command, after giving invalid UserType value we are getting success response

**Actual Behavior** -After giving UserType  value as invalid we are getting success response

**Expected Behavior** - After giving user type value as invalid we have to get INVALID_COMMAND

App used - lock app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT - RPI-4, 8GB RAM
Network - Ble-wifi
Commit -  9493d7b48c410f058b85552e1668b33f858afcac

**TH LOG :**

```
./chip-tool doorlock set-user 0 2 xxx 6451 1 10 3 1 1 --timedInteractionTimeoutMs 1000
[1655212761.421326][19184:19184] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1655212761.422174][19184:19184] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1655212761.422370][19184:19184] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1655212761.422504][19184:19184] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1655212761.422914][19184:19184] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-1WRyp5)
[1655212761.423627][19184:19184] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1655212761.423701][19184:19184] CHIP:DL: NVS set: chip-counters/reboot-count = 108 (0x6C)
[1655212761.424622][19184:19184] CHIP:DL: Got Ethernet interface: eth0
[1655212761.425298][19184:19184] CHIP:DL: Found the primary Ethernet interface:eth0
[1655212761.425970][19184:19184] CHIP:DL: Got WiFi interface: wlan0
[1655212761.426057][19184:19184] CHIP:DL: Failed to reset WiFi statistic counts
[1655212761.426122][19184:19184] CHIP:IN: UDP::Init bind&listen port=0
[1655212761.426279][19184:19184] CHIP:IN: UDP::Init bound to port=35671
[1655212761.426307][19184:19184] CHIP:IN: BLEBase::Init - setting/overriding transport
[1655212761.426328][19184:19184] CHIP:IN: TransportMgr initialized
[1655212761.426418][19184:19184] CHIP:FP: Initializing FabricTable from persistent storage
[1655212761.426722][19184:19184] CHIP:TS: Last Known Good Time: 2022-06-14T07:55:16
[1655212761.426941][19184:19184] CHIP:FP: Loading from storage for fabric index 0x1
[1655212761.427774][19184:19184] CHIP:FP: Loading from storage for fabric index 0x2
[1655212761.428498][19184:19184] CHIP:FP: Loading from storage for fabric index 0x3
[1655212761.435645][19184:19184] CHIP:ZCL: Using ZAP configuration...
[1655212761.439934][19184:19184] CHIP:DL: Avahi client registered
[1655212761.441134][19184:19184] CHIP:CTL: System State Initialized...
[1655212761.441219][19184:19184] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1655212761.441274][19184:19184] CHIP:CTL: Setting attestation nonce to random value
[1655212761.441338][19184:19184] CHIP:CTL: Setting CSR nonce to random value
[1655212761.441471][19184:19184] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1655212761.441501][19184:19184] CHIP:CTL: Setting attestation nonce to random value
[1655212761.441548][19184:19184] CHIP:CTL: Setting CSR nonce to random value
[1655212761.444361][19184:19184] CHIP:CTL: Generating NOC
[1655212761.445202][19184:19184] CHIP:FP: Validating NOC chain
[1655212761.447047][19184:19184] CHIP:FP: NOC chain validation successful
[1655212761.447213][19184:19184] CHIP:FP: Added new fabric at index: 0x1, Initialized: 1
[1655212761.447239][19184:19184] CHIP:FP: Assigned compressed fabric ID: 0xA1CC5128D551FAC7, node ID: 0x000000000001B669
[1655212761.460651][19184:19184] CHIP:FP: Fabric (0x1) persisted to storage. Calling OnFabricPersistedToStorage
[1655212761.460705][19184:19184] CHIP:TS: Last Known Good Time: 2022-06-14T07:55:16
[1655212761.460729][19184:19184] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1655212761.460752][19184:19184] CHIP:TS: Retaining current Last Known Good Time
[1655212761.465452][19184:19184] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0xA1CC5128D551FAC7
[1655212761.465576][19184:19184] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1655212761.465618][19184:19184] CHIP:CTL: Setting attestation nonce to random value
[1655212761.465696][19184:19184] CHIP:CTL: Setting CSR nonce to random value
[1655212761.466559][19184:19184] CHIP:CTL: Generating NOC
[1655212761.467433][19184:19184] CHIP:FP: Validating NOC chain
[1655212761.469345][19184:19184] CHIP:FP: NOC chain validation successful
[1655212761.469540][19184:19184] CHIP:FP: Added new fabric at index: 0x2, Initialized: 1
[1655212761.469570][19184:19184] CHIP:FP: Assigned compressed fabric ID: 0x13173974745587F1, node ID: 0x000000000001B669
[1655212761.481904][19184:19184] CHIP:FP: Fabric (0x2) persisted to storage. Calling OnFabricPersistedToStorage
[1655212761.481960][19184:19184] CHIP:TS: Last Known Good Time: 2022-06-14T07:55:16
[1655212761.481984][19184:19184] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1655212761.482006][19184:19184] CHIP:TS: Retaining current Last Known Good Time
[1655212761.484817][19184:19184] CHIP:CTL: Joined the fabric at index 2. Compressed fabric ID is: 0x13173974745587F1
[1655212761.484957][19184:19184] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1655212761.484998][19184:19184] CHIP:CTL: Setting attestation nonce to random value
[... 234 lines cut ...]
[1655213700.603624][17592:17592] CHIP:DMG: AccessControl: allowed
[1655213700.603673][17592:17592] CHIP:DMG: Received command for Endpoint=1 Cluster=0x0000_0101 Command=0x0000_001A
[1655213700.603752][17592:17592] CHIP:ZCL: [SetUser] Incoming command [endpointId=1,userIndex=2]
[1655213700.603819][17592:17592] CHIP:ZCL: Lock App: LockEndpoint::GetUser [endpoint=1,userIndex=2]
[1655213700.603860][17592:17592] CHIP:ZCL: Found unoccupied user [endpoint=1,adjustedIndex=1]
[1655213700.603901][17592:17592] CHIP:ZCL: Lock App: LockEndpoint::SetUser [endpoint=1,userIndex=2,creator=1,modifier=1,userName="xxx",uniqueId=1933,userStatus=1,userType=10,credentialRule=3,credentials=(nil),totalCredentials=0]
[1655213700.603950][17592:17592] CHIP:ZCL: Successfully set the user [mEndpointId=1,index=2,adjustedIndex=1]
[1655213700.603988][17592:17592] CHIP:ZCL: [createUser] User created [endpointId=1,creatorFabricId=1,userIndex=2,userName="xxx",userUniqueId=0x1933,userStatus=1,userType=10,credentialRule=3,totalCredentials=0]
[1655213700.604090][17592:17592] CHIP:EVL: LogEvent event number: 0x0000000000000002 priority: 1, endpoint id:  0x1 cluster id: 0x0000_0101 event id: 0x4 Sys timestamp: 0x0000000001A80207
[1655213700.604121][17592:17592] CHIP:ZCL: [RemoteLockUserChange] Sent lock user change event [endpointId=1,eventNumber=2,dataType=2,operation=0,nodeId=112233,fabricIndex=1]
[1655213700.604154][17592:17592] CHIP:DMG: ICR moving to [ Preparing]
[1655213700.604184][17592:17592] CHIP:DMG: ICR moving to [AddingComm]
[1655213700.604261][17592:17592] CHIP:DMG: ICR moving to [AddedComma]
[1655213700.604307][17592:17592] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
[1655213700.604339][17592:17592] CHIP:EM: Piggybacking Ack for MessageCounter:225790208 on exchange: 58667r
[1655213700.604400][17592:17592] CHIP:IN: Prepared secure message 0xaaaae2cfd618 to 0x000000000001B669 (1)  of type 0x9 and protocolId (0, 1) on exchange 58667r with MessageCounter:223344284.
[1655213700.604439][17592:17592] CHIP:IN: Sending encrypted msg 0xaaaae2cfd618 with MessageCounter:223344284 to 0x000000000001B669 (1) at monotonic time: 0000000001A80207 msec
[1655213700.604610][17592:17592] CHIP:DMG: ICR moving to [CommandSen]
[1655213700.604637][17592:17592] CHIP:DMG: ICR moving to [AwaitingDe]
[1655213700.606340][17592:17592] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:225790209 on exchange 58667r
[1655213700.606372][17592:17592] CHIP:EM: Found matching exchange: 58667r, Delegate: (nil)
[1655213700.606407][17592:17592] CHIP:EM: Rxd Ack; Removing MessageCounter:223344284 from Retrans Table on exchange 58667r
[1655213700.606431][17592:17592] CHIP:EM: Removed CHIP MessageCounter:223344284 from RetransTable on exchange 58667r

```

bzbarsky-apple, 2022-06-14:
@Morozov-5F do you have time to take a look at this one?  Or any pointers to anyone else working on door locks?

andy31415, 2022-06-14:
The actual command would also be useful for fast reproduction. the command you used for `chip-tool` (or set of commands) would allow developers to quickly reproduce this.

bzbarsky-apple, 2022-06-14:
> The actual command would also be useful for fast reproduction

It's at the top of the first log:
```
./chip-tool doorlock set-user 0 2 xxx 6451 1 10 3 1 1 --timedInteractionTimeoutMs 1000
```
That "10" is out of range of the enum.

sumaky, 2022-06-15:
@bzbarsky-apple 10 is out of range. We expected a failure but seeing success.

Morozov-5F, 2022-06-15:
It is really interesting. The `userType` is provided to cluster code as an `enum class DlUserType`, which is generated straight from `door-lock-cluster.xml` definition. 
I assumed that such parameters range would be checked by the underlying SDK code but it appears to be the opposite. I'm still not sure how to feel about this but anyway it requires a lot of code changes, mostly trivial: we need to add the range checks for each parameter in the commands.

woody-apple, 2022-06-15:
Spec Issue Review: Assigning to @Morozov-5F

bzbarsky-apple, 2022-06-16:
@Morozov-5F Yes, the underlying code does not yet know to do range checks on enums....  Sorry.  :(

Morozov-5F, 2022-06-16:
@bzbarsky-apple no problem, should be a fairly straightforward fix

PR #19813 (2022-07-07): Fix incorrect return codes in door lock cluster
#### Problem
* Fixes #19622
* Fixes #19563

#### Change overview
* Now door lock cluster returns generic failure instead of NOT_FOUND when working with Week Day/Year Day credentials.
* Door lock now checks the enum ranges for Set Credential, Set User and Set Holiday Schedule commands.
* Refactor the door lock cluster a little bit

#### Testing
* Run DL_LockUnlock, DL_Schedules and DL_UsersAndCredentials tests
* Run commands from linked issues to make sure they return appropriate error codes
PR #20433 (2022-07-07): Fix incorrect return codes in door lock cluster
#### Problem
* Fixes #19622
* Fixes #19563

#### Change overview
* Now door lock cluster returns generic failure instead of NOT_FOUND when working with Week Day/Year Day credentials.
* Door lock now checks the enum ranges for Set Credential, Set User and Set Holiday Schedule commands.
* Refactor the door lock cluster a little bit

#### Testing
* Run DL_LockUnlock, DL_Schedules and DL_UsersAndCredentials tests
* Run commands from linked issues to make sure they return appropriate error codes
closed 2022-07-07 by selissia