project-chip/connectedhomeip #19591: Crash: CHIP:  chip::ChipError chip::Controller::ClusterBase::ReadAttribute<chip::Span<char const>, chip::Span<char const> >

aajain-com, 2022-06-14:
#### Problem

Seeing this crash reported during testing on iOS/tvOS. This is seen as recent as TE9 SHA `cfc35951be66a664a6efdadea56d1b8ea6e63e96`.

```
Thread 1 Crashed::   Dispatch queue: com.zigbee.chip.framework.controller.workqueue
0   CHIP                          	       0x102426b30 chip::ChipError chip::Controller::ClusterBase::ReadAttribute, chip::Span >(void*, unsigned int, unsigned int, void (*)(void*, chip::Span), void (*)(void*, chip::ChipError), bool, chip::Optional const&) + 108 (/Library/Caches/com.apple.xbs/Sources/61e3901f-3a8c-438d-94e0-b9d8be91782b/CHIPFramework-58/connectedhomeip/src/controller/CHIPCluster.h:254)
1   CHIP                          	       0x1021e06f4 __58-[CHIPBasic readAttributeVendorNameWithCompletionHandler:]_block_invoke + 108 (/Library/Caches/com.apple.xbs/Sources/61e3901f-3a8c-438d-94e0-b9d8be91782b/CHIPFramework-58/connectedhomeip/src/controller/CHIPCluster.h:228)
2   CHIP                          	       0x1021e06f4 __58-[CHIPBasic readAttributeVendorNameWithCompletionHandler:]_block_invoke + 108 (/Library/Caches/com.apple.xbs/Sources/61e3901f-3a8c-438d-94e0-b9d8be91782b/CHIPFramework-58/connectedhomeip/src/controller/CHIPCluster.h:228)
3   CHIP                          	       0x1023c0984 ___ZN18CHIPCallbackBridgeIPFvPvN4chip4SpanIKcEEEEC2EPU28objcproto17OS_dispatch_queue8NSObjectU13block_pointerFvP11objc_objectP7NSErrorEU13block_pointerFNS1_9ChipErrorEPNS1_8Callback10CancelableESK_ES6_b_block_invoke + 116 (/Library/Caches/com.apple.xbs/Sources/61e3901f-3a8c-438d-94e0-b9d8be91782b/CHIPFramework-58/connectedhomeip/src/darwin/Framework/CHIP/CHIPCallbackBridgeBase_internal.h:45)
4   libdispatch.dylib             	       0x1bd2470f4 _dispatch_client_callout + 20 (/Library/Caches/com.apple.xbs/Sources/libdispatch/src/object.m:560)
5   libdispatch.dylib             	       0x1bd256a60 _dispatch_sync_invoke_and_complete_recurse + 64 (/Library/Caches/com.apple.xbs/Sources/libdispatch/src/queue.c:1023)
6   libdispatch.dylib             	       0x1bd2564e0 _dispatch_sync_f_slow + 172 (/Library/Caches/com.apple.xbs/Sources/libdispatch/src/queue.c:1780)
7   CHIP                          	       0x1023c0734 CHIPCallbackBridge)>::CHIPCallbackBridge(NSObject*, void (objc_object*, NSError*) block_pointer, chip::ChipError (chip::Callback::Cancelable*, chip::Callback::Cancelable*) block_pointer, void (*)(void*, chip::Span), bool) + 528 (/Library/Caches/com.apple.xbs/Sources/61e3901f-3a8c-438d-94e0-b9d8be91782b/CHIPFramework-58/connectedhomeip/src/darwin/Framework/CHIP/CHIPCallbackBridgeBase_internal.h:44)
8   CHIP                          	       0x1021e0624 -[CHIPBasic readAttributeVendorNameWithCompletionHandler:] + 176 (/Library/Caches/com.apple.xbs/Sources/61e3901f-3a8c-438d-94e0-b9d8be91782b/CHIPFramework-58/connectedhomeip/src/darwin/Framework/CHIP/zap-generated/CHIPCallbackBridge_internal.h:693)
9   CoreFoundation                	       0x1bd57e0c4 __invoking___ + 148
10  CoreFoundation                	       0x1bd59b780 -[NSInvocation invoke] + 428 (/Library/Caches/com.apple.xbs/Sources/CoreFoundation/CoreFoundation/ObjC.subproj/NSForwarding.m:3355)
11  CoreFoundation                	       0x1bd5d0f04 -[NSInvocation invokeWithTarget:] + 64 (/Library/Caches/com.apple.xbs/Sources/CoreFoundation/CoreFoundation/ObjC.subproj/NSForwarding.m:3452)
12  CHIPPlugin                    	       0x100e08b04 -[CHIPPluginProtocolOperationManager registerOperation:clientQueue:operationResponseHandler:] + 1572 (/Library/Caches/com.apple.xbs/Sources/7ec73f8c-aa7e-49e1-8258-0e1420b1dd21/CHIPPlugin-113/CHIPPlugin/Utilities/CHIPPluginProtocolOperationManager.m:249)
13  CHIPPlugin                    	       0x100e2fee8 -[CHIPAccessoryServer _readCharacteristicValues:forFailedResponses:atIndexes:device:completionHandler:] + 460 (/Library/Caches/com.apple.xbs/Sources/7ec73f8c-aa7e-49e1-8258-0e1420b1dd21/CHIPPlugin-113/CHIPPlugin/Server/CHIPAccessoryServer.m:1397)
14  CHIPPlugin                    	       0x100e300f8 __103-[CHIPAccessoryServer _readCharacteristicValues:forFailedResponses:atIndexes:device:completionHandler:]_block_invoke + 232 (/Library/Caches/com.apple.xbs/Sources/7ec73f8c-aa7e-49e1-8258-0e1420b1dd21/CHIPPlugin-113/CHIPPlugin/Server/CHIPAccessoryServer.m:1402)
15  libdispatch.dylib             	       0x1bd2455cc _dispatch_call_block_and_release + 32 (/Library/Caches/com.apple.xbs/Sources/libdispatch/src/init.c:1518)
16  libdispatch.dylib             	       0x1bd2470f4 _dispatch_client_callout + 20 (/Library/Caches/com.apple.xbs/Sources/libdispatch/src/object.m:560)
17  libdispatch.dylib             	       0x1bd24e7ac _dispatch_lane_serial_drain + 672 (/Library/Caches/com.apple.xbs/Sources/libdispatch/src/inline_internal.h:2631)
18  libdispatch.dylib             	       0x1bd24f32c _dispatch_lane_invoke + 436 (/Library/Caches/com.apple.xbs/Sources/libdispatch/src/queue.c:3939)
19  libdispatch.dylib             	       0x1bd259ebc _dispatch_workloop_worker_thread + 652 (/Library/Caches/com.apple.xbs/Sources/libdispatch/src/queue.c:6766)
20  libsystem_pthread.dylib       	       0x2370ec040 _pthread_wqthread + 288 (/Library/Caches/com.apple.xbs/Sources/libpthread/src/pthread.c:2611)
21  libsystem_pthread.dylib       	       0x2370ebde0 start_wqthread + 8
```

aajain-com, 2022-07-27:
Marking this as P1 given that this is one of the top hitters.

bzbarsky-apple, 2022-08-02:
It's somewhat likely that https://github.com/project-chip/connectedhomeip/pull/21504 fixed this....

franck-apple, 2022-08-04:
@aajain-com are you able to confirm?

bzbarsky-apple, 2022-08-04:
I don't see how @aajain-com can confirm that a change that he's not able to test yet fixes anything, fwiw... ;)

aajain-com, 2022-08-04:
I can keep an eye on this issue once we have had a chance to try with the potential fix and get back.

andy31415, 2022-08-24:
Removed blocking tag: platform specific fix and there is some suspicion that this may already be fixed.

woody-apple, 2022-08-30:
Resolved with: https://github.com/project-chip/connectedhomeip/pull/21504

closed 2022-08-30 by woody-apple