project-chip/connectedhomeip #19593: chip::CommissioneeDeviceProxy::IsSessionSetupInProgress const

aajain-com, 2022-06-14:
#### Problem

Seeing this crash reported during testing on iOS/tvOS. This is seen as recent as TE9 SHA `cfc35951be66a664a6efdadea56d1b8ea6e63e96`.

```
Thread 10 Crashed::   Dispatch queue: com.zigbee.chip.framework.controller.workqueue
0   CHIP                          	       0x1fc98b538 chip::CommissioneeDeviceProxy::IsSessionSetupInProgress() const + 44 (CHIPFramework/connectedhomeip/src/controller/CommissioneeDeviceProxy.h:202)
1   CHIP                          	       0x1fc98b458 chip::Controller::DeviceCommissioner::Shutdown() + 144 (CHIPFramework/connectedhomeip/src/controller/CHIPDeviceController.cpp:662)
2   CHIP                          	       0x1fc98b458 chip::Controller::DeviceCommissioner::Shutdown() + 144 (CHIPFramework/connectedhomeip/src/controller/CHIPDeviceController.cpp:662)
3   CHIP                          	       0x1fc559e2c __32-[CHIPDeviceController shutdown]_block_invoke + 76 (CHIPFramework/connectedhomeip/src/darwin/Framework/CHIP/CHIPDeviceController.mm:141)
4   libdispatch.dylib             	       0x180cbd5d0 _dispatch_call_block_and_release + 32
5   libdispatch.dylib             	       0x180cbf0f8 _dispatch_client_callout + 20
6   libdispatch.dylib             	       0x180cc67ac _dispatch_lane_serial_drain + 672
7   libdispatch.dylib             	       0x180cc72f8 _dispatch_lane_invoke + 384
8   libdispatch.dylib             	       0x180cd1ebc _dispatch_workloop_worker_thread + 652
9   libsystem_pthread.dylib       	       0x1fa3c90a8 _pthread_wqthread + 288
10  libsystem_pthread.dylib       	       0x1fa3c8e48 start_wqthread + 8
```

 #### Proposed Solution

bzbarsky-apple, 2022-08-02:
> This is seen as recent as TE9 SHA cfc35951be66a664a6efdadea56d1b8ea6e63e96.
...
> 0   CHIP                          	       0x1fc98b538 chip::CommissioneeDeviceProxy::IsSessionSetupInProgress() const + 44 (CHIPFramework/connectedhomeip/src/controller/CommissioneeDeviceProxy.h:202)

There is no line 202 in that file on that SHA...  But presumably `mDeviceInPASEEstablishment` was garbage, since we do null-check it on that SHA and `IsSessionSetupInProgress` is a simple member integer value check with no way to crash as long as `this` is valid.

Since the TE9 SHA there have been several fixes to prevent dangling pointers and use-after-free there, so I would _guess_ this is fixed on tip, but it's hard to say for sure...

aajain-com, 2022-08-03:
@bzbarsky-apple I looked at the crash reporting tool and I haven't seen this issue since we integrated with pre SVE SHA.

bzbarsky-apple, 2022-08-03:
@aajain-com Sounds good.  Let's keep an eye on this to see if it does resurface, but if not at some point it feels like we should just close it.

woody-apple, 2022-08-30:
This is no longer showing up in crashes, closing

closed 2022-08-30 by woody-apple