project-chip/connectedhomeip #19594: [Crash] chip::Controller::DeviceCommissioner::ContinueCommissioningAfterDeviceAttestationFailure

aajain-com, 2022-06-14:
#### Problem

Seeing this crash reported during testing on iOS/tvOS. This is seen as recent as TE9 SHA `cfc35951be66a664a6efdadea56d1b8ea6e63e96`.

```
Thread 3 Crashed::   Dispatch queue: com.zigbee.chip.framework.controller.workqueue
0   CHIP                          	       0x108128f28 chip::Controller::DeviceCommissioner::ContinueCommissioningAfterDeviceAttestationFailure(chip::DeviceProxy*, chip::Credentials::AttestationVerificationResult) + 192 (connectedhomeip/src/controller/CHIPDeviceController.cpp:727)
1   CHIP                          	       0x107cc1260 __83-[CHIPDeviceController continueCommissioningDevice:ignoreAttestationFailure:error:]_block_invoke + 96 (connectedhomeip/src/darwin/Framework/CHIP/CHIPDeviceController.mm:435)
2   CHIP                          	       0x107cc1260 __83-[CHIPDeviceController continueCommissioningDevice:ignoreAttestationFailure:error:]_block_invoke + 96 (connectedhomeip/src/darwin/Framework/CHIP/CHIPDeviceController.mm:435)
3   libdispatch.dylib             	       0x19be730f4 _dispatch_client_callout + 20
4   libdispatch.dylib             	       0x19be8262c _dispatch_lane_barrier_sync_invoke_and_complete + 56
5   CHIP                          	       0x107cc115c -[CHIPDeviceController continueCommissioningDevice:ignoreAttestationFailure:error:] + 272 (connectedhomeip/src/darwin/Framework/CHIP/CHIPDeviceController.mm:428)
6   CHIPPlugin                    	       0x107004024 __71-[CHIPPluginAttestationStatus deviceAttestation:failedForDevice:error:]_block_invoke + 332
7   libdispatch.dylib             	       0x19be715cc _dispatch_call_block_and_release + 32
8   libdispatch.dylib             	       0x19be730f4 _dispatch_client_callout + 20
9   libdispatch.dylib             	       0x19be7a7ac _dispatch_lane_serial_drain + 672
10  libdispatch.dylib             	       0x19be7b32c _dispatch_lane_invoke + 436
11  libdispatch.dylib             	       0x19be85ebc _dispatch_workloop_worker_thread + 652
12  libsystem_pthread.dylib       	       0x216416040 _pthread_wqthread + 288
13  libsystem_pthread.dylib       	       0x216415de0 start_wqthread + 8

```

bzbarsky-apple, 2022-07-14:
Crashing line is:
```
    CommissioneeDeviceProxy * commissioneeDevice = FindCommissioneeDevice(device->GetDeviceId());
```

I wonder whether the controller is shut down already or something...  But we're supposed to have checks for that....

woody-apple, 2022-08-30:
This is no longer showing up in crashes, closing

closed 2022-08-30 by woody-apple