project-chip/connectedhomeip #19604: Doorlock clear credential command and get credential command  return failure when called after set credential command

Ashwinigrl, 2022-06-15:
**Problem** -when we try to send Clear Credential Command after setting credential we are seeing failure 

**sequence of  steps** 
1. clear credential =./chip-tool doorlock clear-credential  '{ "credentialType" : 1 , "credentialIndex" : 1 }'  1 1 --timedInteractionTimeoutMs 1000   (Response=success)
2. set credential =./chip-tool doorlock set-credential 0 '{ "credentialType" : 1 , "credentialIndex" : 1 }' 123456 null 0 0 1 1 --timedInteractionTimeoutMs 1000   (Response=success)
3. get credential = ./chip-tool doorlock get-credential-status '{ "credentialType" : 1 , "credentialIndex" : 1 }'  1 1 --timedInteractionTimeoutMs 1000   (Response=failure)
4. clear credential= ./chip-tool doorlock clear-credential  '{ "credentialType" : 1 , "credentialIndex" : 1 }'  1 1 --timedInteractionTimeoutMs 1000   (Response=failure)

**Expected Behavior** - in step 3 and 4 status should be success 0x00 (Success)

App used - lock app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT - RPI-4, 8GB RAM
Network - Ble-wifi
Commit - https://github.com/project-chip/connectedhomeip/commit/9493d7b48c410f058b85552e1668b33f858afcac

bzbarsky-apple, 2022-06-15:
From the server log:
```
[1655305543142] [3213:12083751] CHIP: [ZCL] [GetCredentialStatus] Database possibly corrupted - credential exists without user assigned [endpointId=1,credentialType=1,credentialIndex=1]
```
The issue is that this command:
```
/chip-tool doorlock set-credential 0 '{ "credentialType" : 1 , "credentialIndex" : 1 }' 123456 null 0 0 1 1 --timedInteractionTimeoutMs 1000
```
is passing in 0 for UserStatus and 0 for UserType, which correspond to "Available" (i.e. "does not exist") and "UnrestrictedUser".

It seems like that should perhaps error out at that point, instead of claiming success and creating a credential attached to a non-existent user?  @DanM-AssaAbloy @Morozov-5F @RyanLindemanCAE

But if it _is_ ok to have a credential attached to a non-existent user, then I'd expect "Get Credential Status" to succeed for such a credential...

Morozov-5F, 2022-06-15:
@bzbarsky-apple yeah, that was definitely overlooked in the implementation. Should be an easy fix.

closed 2022-06-30 by Ashwinigrl