project-chip/connectedhomeip #19610: TC-GR-2.2 - Groups Cluster - RemoveallGroups command not working

sethunk, 2022-06-15:
**1.Problem** - when RemoveAllGroups command is sent from TH to DUT , the group still exists in the GroupTable

**2.Sequence of steps** 
**removeallgroups** - ./chip-tool groups remove-all-groups 1 0
**grouptable** - ./chip-tool groupkeymanagement read group-table 1 0

**3.Test Environment**

**App used** - allclusters app
**Platform** - Chip-tool - RPI-4, 8GB RAM
DUT - RPI-4, 8GB RAM
**Network** - Ble-wifi
**Commit** - 9493d7b48c410f058b85552e1668b33f858afcac

**4.Logs**

[DUT_Logs.txt](https://github.com/project-chip/connectedhomeip/files/8907352/DUT_Logs.txt)
[TH_Logs.txt](https://github.com/project-chip/connectedhomeip/files/8907353/TH_Logs.txt)

bzbarsky-apple, 2022-06-15:
This looks like an issue similar to the one #19221 fixed.  It looks like the two-arg form of `GroupDataProviderImpl::RemoveEndpoint` duplicates a bunch of the code from the three-arg form, and while the latter was fixed the former was not.

@rcasallas-silabs Could you take a look, please?  Ideally the impls would just be shared as much as possible, to avoid these problems....

woody-apple, 2022-06-15:
Cert Blocker Review: Tagging Cert blocker, keeping assigned to @rcasallas-silabs