project-chip/connectedhomeip #19611: TC-GR-2.3 Groups Cluster -  AddGroupIfIdentifying command not working

sethunk, 2022-06-15:
**1.Problem** - 1. When TH adds the group using  AddGroupIfIdentifying command, the group doesn't exists in GroupTable
		  2. when TH adds n+1 group using  AddGroupIfIdentifying command, we are unable to get RESOUCE_EXHAUSTED response 

**2.Sequence of steps**
 **AddGroupIfIdentifying** -  ./chip-tool groups  add-group-if-identifying 0x0006 gp6 1 0
 **GroupTable**  -  ./chip-tool groupkeymanagement read group-table 1 0

	
**3.Test Environment**
	
**App used** - allclusters app
**Platform** - Chip-tool - RPI-4, 8GB RAM
           DUT - RPI - RPI-4, 8GB RAM
**Network** - Ble-wifi
**Commit** - 9493d7b48c410f058b85552e1668b33f858afcac

**4.Logs**
[DUT_Logs_15.06.2022.txt](https://github.com/project-chip/connectedhomeip/files/8907445/DUT_Logs_15.06.2022.txt)
[TH_Logs_15.06.2022.txt](https://github.com/project-chip/connectedhomeip/files/8907446/TH_Logs_15.06.2022.txt)

bzbarsky-apple, 2022-06-15:
AddGroupIfIdentifying is not part of Matter.

bzbarsky-apple, 2022-06-15:
Looking at the test plan at https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/Groups.adoc#323-tc-g-23-commands---getgroupmembership-addgroupifidentifying-dut-server it looks like the pre-conditions do _not_ have "DUT is identifying" as a pre-condition, and nothing in the test plan tells it to start identifying.  So it sounds like the test plan is buggy and needs fixing.

woody-apple, 2022-06-15:
Cert Blocker Review: @bzbarsky-apple will file a test plan issue, so closing this.

closed 2022-06-15 by woody-apple