project-chip/connectedhomeip #19613: Groups Cluster- Issue in NameSupport attribute value

sethunk, 2022-06-15:
**1.Problem** -  NameSupport attribute value is 128 when the FeatureMap value is 0 

**2.Sequence of steps**
Featuremap - ./chip-tool groups read feature-map 1 0
NameSupport - ./chip-tool groups read name-support 1 0

**3.Test Environment**

App used - allclusters app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Network - Ble-wifi
Commit - 9493d7b48c410f058b85552e1668b33f858afcac

**4.Logs**
[DUT_Feature.txt](https://github.com/project-chip/connectedhomeip/files/8907705/DUT_Feature.txt)
[TH_Feature.txt](https://github.com/project-chip/connectedhomeip/files/8907708/TH_Feature.txt)

bzbarsky-apple, 2022-06-15:
Looks like the groups cluster impl forces a particular NameSupport value at startup but does not touch the FeatureMap.

@rcasallas-silabs

woody-apple, 2022-06-15:
Cert Blocker Review: Assigning to @rcasallas-silabs to take a look at this. Thanks!

franck-apple, 2022-06-21:
Looks like we can close this one, as the duplicate https://github.com/project-chip/connectedhomeip/issues/19765 was closed as completed.
Please re-open if it's not the case.

bzbarsky-apple, 2022-06-22:
No, this does not make any sense.  The duplicate was closed as a duplicate _of this issue_.

andy31415, 2022-07-07:
Trying to take a look ... I don't know much about groups, but it does look like this should be testable easily enough.

andy31415, 2022-07-07:
> please see [CHIP-Specifications/connectedhomeip-spec#5286](https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5286) - NameSupport is there for legacy reasons (remember Matter and Zigbee share the work from Data Model, cannot ignore the heritage) and needs to be linked to the feature bit which is leading.

@leorozendaal 
SDK feedback and clean slate reader: this is redundant and confusing. We will end up some clients using one feature and others use a different attribute. "backwards compatibility" should not be a justification for a 1.0 protocol as there is no "backward". 

I understand a mandate of 'when in doubt, make migration from zigbee easy' however this is an awkward/redundant implementation for the sake of 'backwards compatibiltiy' and from that perspective I believe it is wrong.

PR #20458 (2022-07-09): Mark 'group names supported' in the groups cluster feature map
#### Problem
Group name support must be in sync between feature map and namesupport. Only namesupport used to be set.

Fixes #19613

#### Change overview
- Define the feature map bitflags in xml
- Use the feature map bitflags in groups server
- minor constant rename for namesupport flags

#### Testing
Compiled all-clusters-app on linux. Got:

```
❯ ./out/linux-x64-chip-tool/chip-tool groups read name-support 1 0 | grep -A1 'Endpoint: 0 Cluster:'
[1657221972.727541][3275513:3275519] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0004 Attribu [... cut]
PR #20527 (2022-07-09): Mark 'group names supported' in the groups cluster feature map
#### Problem
Group name support must be in sync between feature map and namesupport. Only namesupport used to be set.

Fixes #19613

#### Change overview
- Define the feature map bitflags in xml
- Use the feature map bitflags in groups server
- minor constant rename for namesupport flags

#### Testing
Compiled all-clusters-app on linux. Got:

```
❯ ./out/linux-x64-chip-tool/chip-tool groups read name-support 1 0 | grep -A1 'Endpoint: 0 Cluster:'
[1657221972.727541][3275513:3275519] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0004 Attribu [... cut]
closed 2022-07-09 by andy31415