project-chip/connectedhomeip #19620: Test_TC_CGEN_2_1.yaml BasicCommissioningInfo should not use specific hardcoded values

tehampson, 2022-06-15:
#### Problem
After https://github.com/project-chip/connectedhomeip/pull/19582 this xml looks like
```
          value:
              {
                  FailSafeExpiryLengthSeconds: 60,
                  MaxCumulativeFailsafeSeconds: 900,
              }
```

While these are the default values they are not the required values and cert testing should just make sure that the values are there

 #### Proposed Solution
<suggested fix, suggested enhanceme<suggested fix, suggested enhancement>nt>

bzbarsky-apple, 2022-06-15:
@kowsisoundhar12 please take a look?

raju-apple, 2022-06-16:
Currently looks like we don't have a way to check the range within a struct value in yaml (https://github.com/project-chip/connectedhomeip/issues/17997) , But for now we can put up a manual user prompt just for this step for the user to validate its within the range and not checking a specific value. @kvikrambhat Can you please put the PR for this modified script once its up please ?

closed 2022-06-22 by franck-apple