project-chip/connectedhomeip #19622: Doorlock Set Week Day Schedule, Get Week Day Schedule,Clear Week Day Schedule command and Set Year Day Schedule, Get Year Day Schedule, Clear Year Day Schedule command return (NOT_FOUND)

Ashwinigrl, 2022-06-15:
**Problem** -when we try to send Set Week Day Schedule, Get Week Day Schedule, Clear Week Day Schedule command and Set Year Day Schedule, Get Year Day Schedule, Clear Year Day Schedule command we are getting (NOT_FOUND)

**sequence of steps**

Set Week Day Schedule =./chip-tool doorlock set-week-day-schedule 2 1 2  15 45 16 55 1 1 (Response=NOT_FOUND)
Get Week Day Schedule =./chip-tool doorlock get-week-day-schedule 2 1 1 1  (Response=NOT_FOUND)
Clear Week Day Schedule =./chip-tool doorlock clear-week-day-schedule 2 1 1 1  (Response=NOT_FOUND)

Set Year Day Schedule =./chip-tool doorlock set-year-day-schedule 1 1 960 1980 1 1 (Response=NOT_FOUND)
Get Year Day Schedule =./chip-tool doorlock get-year-day-schedule 1 1 1 1   (Response=NOT_FOUND)
Clear Year Day Schedule =./chip-tool doorlock clear-year-day-schedule 1 1 1 1  ( (Response=NOT_FOUND)

**Expected Behavior -** 
[DOOR LOCK Logs.txt](https://github.com/project-chip/connectedhomeip/files/8909985/DOOR.LOCK.Logs.txt)
 status should be success 0x00 (Success)

**Additional information**

1654673540.052851][3218:3223] CHIP:TOO:   AcceptedCommandList: 18 entries
[1654673540.052894][3218:3223] CHIP:TOO:     [1]: 0
[1654673540.052929][3218:3223] CHIP:TOO:     [2]: 1
[1654673540.052963][3218:3223] CHIP:TOO:     [3]: 3
**[1654673540.052997][3218:3223] CHIP:TOO:     [4]: 11**
****[1654673540.053031][3218:3223] CHIP:TOO:     [5]: 12**
**[1654673540.053064][3218:3223] CHIP:TOO:     [6]: 13****
**[1654673540.053097][3218:3223] CHIP:TOO:     [7]: 14**
**[1654673540.053129][3218:3223] CHIP:TOO:     [8]: 15**
**[1654673540.053162][3218:3223] CHIP:TOO:     [9]: 16**
[1654673540.053195][3218:3223] CHIP:TOO:     [10]: 17
[1654673540.053228][3218:3223] CHIP:TOO:     [11]: 18
[1654673540.053262][3218:3223] CHIP:TOO:     [12]: 19
[1654673540.053294][3218:3223] CHIP:TOO:     [13]: 26
[1654673540.053327][3218:3223] CHIP:TOO:     [14]: 27
[1654673540.053360][3218:3223] CHIP:TOO:     [15]: 29
[1654673540.053392][3218:3223] CHIP:TOO:     [16]: 34
[1654673540.053425][3218:3223] CHIP:TOO:     [17]: 36
[1654673540.053458][3218:3223] CHIP:TOO:     [18]: 38

App used - lock app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT - RPI-4, 8GB RAM
Network - Ble-wifi
Commit - https://github.com/project-chip/connectedhomeip/commit/9493d7b48c410f058b85552e1668b33f858afcac

bzbarsky-apple, 2022-06-15:
> Set Week Day Schedule =./chip-tool doorlock set-week-day-schedule 2 1 2 15 45 16 55 1 1 (Response=NOT_FOUND)

@Ashwinigrl Is there an actual user at user index 1?  What commands were run to set up the user before running these schedule commands?

sumaky, 2022-06-16:
> > Set Week Day Schedule =./chip-tool doorlock set-week-day-schedule 2 1 2 15 45 16 55 1 1 (Response=NOT_FOUND)
> 
> @Ashwinigrl Is there an actual user at user index 1? What commands were run to set up the user before running these schedule commands?

@bzbarsky-apple This is an issue with test plan as well.
 https://github.com/CHIP-Specifications/chip-test-plans/issues/1537

bzbarsky-apple, 2022-06-17:
@Morozov-5F It looks like the spec will not change here, so we can't return NOT_FOUND in the SDK.  Need to return one of the spec-allowed values.

Morozov-5F, 2022-06-19:
@bzbarsky-apple but this case is not covered by the spec. Should I return generic error in that case?

bzbarsky-apple, 2022-06-21:
I think generic FAILURE may be our best option here, given what the spec says right now....  Worth checking with Dan Matosian how he expects this to work?

woody-apple, 2022-06-21:
Cert Blocker Review: Making this a cert blocker, we should do something reasonable here in the SDK.

bzbarsky-apple, 2022-06-28:
I don't see how this can possibly be passing, given that the SDK does not implement the spec as far as I know...

Survensa, 2022-06-29:
@bzbarsky-apple 

Re-verified with latest SHA

- > Commit used : 81c7f2aacebbe5bb304a62ecdbeaff9220d72c69
- > Chip-tool : RPI 
- > DUT : Door lock app
- > Network : IP (ethernet)

### Execution LOG :

1. - Week day shedule belongs to TC-DRLK-2.5 test case : [TC-DRLK-2.5 LOG](https://github.com/project-chip/connectedhomeip/files/9010205/TC-DRLK-2.5.LOG)
2. 
3. - Year day shedule belongs to TC-DRLK-2.7 test case : [TC-DRLK-2.7 LOG](https://github.com/project-chip/connectedhomeip/files/9010178/TC-DRLK-2.7.LOG)

bzbarsky-apple, 2022-06-29:
@Survensa Is that running the updated test plan, or the old one?  The old one was broken, as far as I can tell, but highlighted an SDK issue.  This is tracking that SDK issue.

The log show a `set-user` command, so that looks like the updated test plan.

bzbarsky-apple, 2022-06-29:
@Survensa I updated the summary to make it clear what the issue is about.

mykrupp, 2022-06-30:
The test spec is not very clear. Anything that involves setting credentials, or holidays requires setting a user first. 

Some test cases state something like: "Assume everything is set up properly to be able to set credentials and schedules". But these should explicitly mention that a user needs to be first, and setting a user needs to be added to the test plan in some cases.

sumaky, 2022-06-30:
@mykrupp This issue is mentioning that the steps are executed when there is no user setup
The command returns NOT_FOUND. It means user not found. So this is not an issue

mykrupp, 2022-06-30:
Ah I understand, thanks for clarifying

bzbarsky-apple, 2022-07-01:
Please stop closing this issue.  NOT_FOUND is **not a valid return per spec** here.  That's what the issue is about.

PR #19813 (2022-07-07): Fix incorrect return codes in door lock cluster
#### Problem
* Fixes #19622
* Fixes #19563

#### Change overview
* Now door lock cluster returns generic failure instead of NOT_FOUND when working with Week Day/Year Day credentials.
* Door lock now checks the enum ranges for Set Credential, Set User and Set Holiday Schedule commands.
* Refactor the door lock cluster a little bit

#### Testing
* Run DL_LockUnlock, DL_Schedules and DL_UsersAndCredentials tests
* Run commands from linked issues to make sure they return appropriate error codes
PR #20433 (2022-07-07): Fix incorrect return codes in door lock cluster
#### Problem
* Fixes #19622
* Fixes #19563

#### Change overview
* Now door lock cluster returns generic failure instead of NOT_FOUND when working with Week Day/Year Day credentials.
* Door lock now checks the enum ranges for Set Credential, Set User and Set Holiday Schedule commands.
* Refactor the door lock cluster a little bit

#### Testing
* Run DL_LockUnlock, DL_Schedules and DL_UsersAndCredentials tests
* Run commands from linked issues to make sure they return appropriate error codes
closed 2022-07-07 by selissia