project-chip/connectedhomeip #19624: Group messaging on EFR32

dtodor, 2022-06-15:
Group messages are not delivered to EFR32 devices.  According to the logs, the EFR32 device joins the group but does not receive any messages. The problem might be related to the configuration of the OpenThread Border Router.

The initial discussion was started in this thread: #19608.

mkardous-silabs, 2022-06-15:
@dtodor
Group commands work on efr32 MG12 and MG24. 
There might be an issue with the commands you are using or the setup as you mentionned.

Did you send the same commands mentionned in the referenced ticket?

dtodor, 2022-06-15:
We are using EFR32 MG12. Yes, we are sending the same commands. The OTBR sees the multicasts, but no packets are received by the device.

mkardous-silabs, 2022-06-15:
I was able to validate the issue is not with the device but with the commands used to configure the groups.
This issue can be closed. I will add the commands I used in the associated ticket.

jepenven-silabs, 2022-06-16:
Closing since working on our side. Verified by @mkardous-silabs . Walk-trough can be seen here : https://github.com/project-chip/connectedhomeip/issues/19608#issuecomment-1156931777

closed 2022-06-16 by jepenven-silabs

dtodor, 2022-06-20:
I would assume that when the device joins the multicast group, I would be at least able to see the `UDP Message Received packet nb` logs. Is this a correct assumption?

dtodor, 2022-06-22:
We are testing using the `lighting-app` example built using the following commands:
```
# Full device
gn gen out/normal '--args=efr32_board="BRD4161A"'
ninja -C out/normal
```

We've also added the following OpenThread configs to `args.gni`, although not sure whether these are needed at all:
```
openthread_enable_core_config_args = true
openthread_config_thread_version = "1.2"
```

jepenven-silabs, 2022-06-23:
@dtodor So the way the OTBR works with an end device for multicast is subscription based. The device needs to be part of the thread network and then you need to add a group to it. Make sure both the chip-tool and the end-device are on the same commit as the building of the Multicast address changed recently and mismatch between chip-tool and the end-device in commit hash might result in the issue you've just described. The OTBR won't forward any multicast address to any devices unless at least one device is subscribed to it.

Now For debugging all of this, you'll need Wireshark or some way to actually sniff IPV6 packets.

Easiest setup would be to run these two commands with wireshark running
`chip-tool groups add-group 0x0101 Light <nodeId> 1`
`chip-tool onoff on 0xffffffffffff0101 1`

These commands will fails for multiples reasons on the device side. Among them are : 
- Wrong Key sync policy
- No key set on the end device
- No ACL set on the end device. 

What you should see however is the UDP log on the EFR32 device, on top of seeing the IPV6 packet in Wireshark.
The on-off command can be send from any device connected to the same network as the OTBR.
However if you're face with this log after executing the command with chip-tool

`[1655997317210] [17669:2893819] CHIP: [TOO] Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417:`

That's is a networking issue with your network. You'll need to investigate why the IPV6 routing isn't working on it .

dtodor, 2022-06-23:
But I would assume that pinging the device through the multicast address is a good indicator that the group subscription (MLR, etc.) has taken place and that the OTBR, especially the BBR part of it, are properly working. Right?

dtodor, 2022-07-14:
So after some more debugging, the UDP packets do get actually delivered to the EFR32 board, but the checksum verification fails, i.e.:
```
#ifndef FUZZING_BUILD_MODE_UNSAFE_FOR_PRODUCTION
    SuccessOrExit(error = Checksum::VerifyMessageChecksum(aMessage, aMessageInfo, kProtoUdp));
#endif
```

I've tried to read and compare the contents of the message with the message as captured by the sniffer:
```
    uint8_t buffer[2028];
    auto    read = aMessage.ReadBytes(0, buffer, 2048);
    printf("*********** OT Udp::HandleMessage - message (bytes %d):\n", read);
    for (int i = 0; i < read; i++)
    {
        printf("%02X", buffer[i]);
    }
    printf("\n");
```

Received message:
```
???: 60 0F 86 97 00 4A 11 3F

Source: FD 11 BF 7D D9 94 79 63 00 A3 2A FF FE 9C 3F E3 

Destination: FF 35 00 40 FD 00 00 00 00 00 00 00 01 00 03 E8 

Source port: DA 78 

Destination port: 15 A4

???: 004A

Checksum: 7B 64

Application: 06 E7 B6 81 2B 23 00 00 01 00 00 00 00 00 00 00 E8 03 5D 9C 0B AF 9B 5B 43 1F 4B 71 F7 7F 91 0C C3 FE EF 53 54 F8 BE D0 6D 3A D8 F2 EE 7B DB 21 60 C9 F8 7A 3C 96 64 A3 5E 41 E3 59 3F AE 53 7F 7E BA
```

Captured message using sniffer:
```
???: F8 7D 59 98 EC 8E 6D FF FF 00 30 0D FB 11 00 00 01 C0 A2 16 07 7E 5A 3C 90 EA 33 88 4A 88 17 03 00 00 FC E1 06 6D 04 40 2D 30 00 EE 6C 08 0F 86 97 3F 

Source: FD 11 BF 7D D9 94 79 63 00 A3 2A FF FE 9C 3F E3 

Destination: FF 35 00 40 FD 00 00 00 00 00 00 00 01 00 03 E8 

???: F0 

Source port: DA 78 

Destination port: 15 A4 

Checksum: 7B 64

Application: 06 E7 B6 81 2B 23 00 00 01 00 00 00 00 00 00 00 E8 03 5D 9C 0B AF 9B 5B 43 1F 4B 71 F7 7F 91 0C C3 FE EF 53 54 F8 BE D0 6D 3A D8 F2 EE 7B DB 21 60 C9 F8 7A 3C 96 64 A3 5E 41 E3 59 3F AE 53 7F 7E BA
```

@jepenven-silabs Any ideas what might cause the checksum verification to fail?

dtodor, 2022-07-14:
After disabling the checksum verification and some fixes in our code, group messaging works!!! Could it be that everyone is building with `FUZZING_BUILD_MODE_UNSAFE_FOR_PRODUCTION` defined?