project-chip/connectedhomeip #19638: [pre-sve]tc-OPCREDS-1.2 Step 6

chrisbeach59, 2022-06-15:
Command 12 (AddTrustedRootCertificate) is mandatory and missing
Have Verified it is checked in ZAP GUI.
This passed in the week 1 sha

./chip-tool operationalcredentials read accepted-command-list 1111 0

 [1655319284.941700][2979:2984] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_003E Attribute 0x0000_FFF9 DataVersion: 70461609
[1655319284.941789][2979:2984] CHIP:TOO:   AcceptedCommandList: 8 entries
[1655319284.941823][2979:2984] CHIP:TOO:     [1]: 0
[1655319284.941849][2979:2984] CHIP:TOO:     [2]: 2
[1655319284.941873][2979:2984] CHIP:TOO:     [3]: 4
[1655319284.941898][2979:2984] CHIP:TOO:     [4]: 6
[1655319284.941922][2979:2984] CHIP:TOO:     [5]: 7
[1655319284.941946][2979:2984] CHIP:TOO:     [6]: 9
[1655319284.942015][2979:2984] CHIP:TOO:     [7]: 10
[1655319284.942043][2979:2984] CHIP:TOO:     [8]: 11

andy31415, 2022-06-15:
@chrisbeach59 - what app, platform was this?
SHA may also be useful for whoever does not directly know or follow pre-sve shas.

https://github.com/project-chip/connectedhomeip/blob/master/docs/BUG_REPORT.md#writing-an-effective-bug-report

bzbarsky-apple, 2022-06-16:
AddTrustedRootCertificate is command 11 (0x0b in the spec).

Command 12 was RemoveTrustedRootCertificate, which has been removed from the spec and the SDK.

That said, this is the second issue I see that thinks AddTrustedRootCertificate is command 12; see also #19548.

@chrisbeach59 was there something in the test script that claimed AddTrustedRootCertificate is command 12?  Or in the test plan?  I am not sure what test script you are using and I can't find tc-OPCREDS-1.2 in the test plans at all.....

closed 2022-06-16 by bzbarsky-apple

bzbarsky-apple, 2022-06-16:
@cjandhyala @naveenkommareddi can we get the verification steps here fixed?