project-chip/connectedhomeip #19639: [pre-sve]TC-BSINFO-2.2 Step 1

chrisbeach59, 2022-06-15:
TC-BINFO-2.2 Step 1 - Expected the Event number be 0

./chip-tool basic read-event start-up 1111 0

My result:
1655320431.632359][3086:3092] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0028 Event 0x0000_0000
[1655320431.632427][3086:3092] CHIP:TOO: 	 Event number: 262144
[1655320431.632484][3086:3092] CHIP:TOO: 	 Priority: Critical
[1655320431.632538][3086:3092] CHIP:TOO: 	 Timestamp: 19056
[1655320431.632664][3086:3092] CHIP:TOO:   StartUp: {
[1655320431.632766][3086:3092] CHIP:TOO:     SoftwareVersion: 0
[1655320431.632827][3086:3092] CHIP:TOO:    }

bzbarsky-apple, 2022-06-16:
> TC-BINFO-2.2 Step 1 - Expected the Event number be 0

Why?  Event numbers are (1) persisted across reboots and (2) may increase quite quickly if the node reboots a lot; they go up by about 1000 every reboot in the default SDK config.  That's why it's a 64-bit number, so even with that it won't roll over in practice.

The test plan at https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicemanagement.adoc#422-tc-binfo-22-events-dut-server says nothing about the event number being 0.  @chrisbeach59 does the test script?

closed 2022-06-16 by bzbarsky-apple

chrisbeach59, 2022-06-16:
@bzbarsky-apple The verification steps showed it as 0.

bzbarsky-apple, 2022-06-16:
@chrisbeach59 Thank you!