project-chip/connectedhomeip #19642: pre-sve [TC-MF-1.21][TC-MF-1.22]

Shadi-Silabs, 2022-06-15:
#### Not able to commission the 5th fabric for both tests.
the logs are all below:
[TC-MF-1.21.txt](https://github.com/project-chip/connectedhomeip/files/8912882/TC-MF-1.21.txt)
[TC-MF-1.21-DUT.txt](https://github.com/project-chip/connectedhomeip/files/8912883/TC-MF-1.21-DUT.txt)
[TC-MF-1.22.txt](https://github.com/project-chip/connectedhomeip/files/8912884/TC-MF-1.22.txt)
[TC-MF-1.22-DUT.txt](https://github.com/project-chip/connectedhomeip/files/8912885/TC-MF-1.22-DUT.txt)

bzbarsky-apple, 2022-06-16:
[TC-MF-1.22-DUT.txt](https://github.com/project-chip/connectedhomeip/files/8912885/TC-MF-1.22-DUT.txt) is an empty file, so not much to say about that one....

bzbarsky-apple, 2022-06-16:
OK, so in the 1.22 DUT log I see:
```
00> <info  > [DL] advertising srp service: 376851A89CC29C75._matterc._udp
00> <info  > [DL] removing srp service: 1A461415045D135D._matterc._udp
00> <info  > [ZCL] Commissioning window is now open
...
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
```
and again the TH log does not show `376851A89CC29C75` anywhere.  So seems like the same issue...

Damian-Nordic, 2022-06-22:
If you get "Timed out waiting on server response" error, it's worth checking OTBR logs as well. In our testing we only encountered "Parsing operation failed" error when sending big SRP updates. https://github.com/project-chip/connectedhomeip/pull/19803, merged yesterday, should reduce the size of our SRP updates, and https://github.com/openthread/openthread/pull/7776 in OpenThread should help process big SRP updates anyway.

closed 2022-07-05 by Shadi-Silabs