project-chip/connectedhomeip #19653: CASESession.cpp:285: CHIP Error 0x00000032: Timeout

Leo2442926161, 2022-06-16:
Dear all,

I am pairing efr32-light-switch to fabric, but sometimes, I met below error, could you take a look? the chip_tool and efr32-light-switch log are attached.

CHIP:TOO: Run command failure: ../../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:285: CHIP Error 0x00000032: Timeout

[efr32-light-switch pairing failed.txt](https://github.com/project-chip/connectedhomeip/files/8914890/efr32-light-switch.pairing.failed.txt)
[efr32-light-switch pairing failed-chip tool.txt](https://github.com/project-chip/connectedhomeip/files/8914891/efr32-light-switch.pairing.failed-chip.tool.txt)

bzbarsky-apple, 2022-06-16:
It's hard to make sense of the server log when there are no timestamps of any sort (even boot-relative) and no logging of incoming payloads, but from the chip-tool log, here is where we finished setting up PASE:
```
[1655342743.385566][52157:52162] CHIP:TOO: Pairing Success
```
Here's where we discovered the operational address near the end of commissioning:
```
[1655342750.785278][52157:52162] CHIP:CTL: Updating device address to UDP:[fd3d:776c:a6df:29fc:d18f:97ad:f844:bfaf]:5540 while in state 1
```
Here's where we sent the Sigma1 message for sending CommissioningComplete:
```
[1655342750.785667][52157:52162] CHIP:SC: Sent Sigma1 msg
```
and we never got a Sigma2 back.

Now in the EFR-side log I am seeing:
```
<detail> [DL] OpenThread State Changed (Flags: 0x00000001)
<detail> [DL]    Thread Unicast Addresses:
<detail> [DL]         fd3d:776c:a6df:29fc:44ac:8665:6f5d:3708/64 valid preferred
<detail> [DL]         fd7f:a92c:bbb8:293:0:ff:fe00:1804/64 valid rloc
<detail> [DL]         fd7f:a92c:bbb8:293:e860:55a5:6571:4650/64 valid
<detail> [DL]         fe80:0:0:0:2061:d24f:b2ea:fb60/64 valid preferred
```
none of which match the address discovered above.  

Is it possible that the same node id (1 in this case) was used to commission multiple different devices which had different IP addresses?

Leo2442926161, 2022-06-20:
@bzbarsky-apple  thanks very much for your help.

I tried it couple times again, but it fails all the time. and I tried remove /tmr/chip*, it's suppose to clean all of chip cache to update the unicast address.

at this time, I notice this address exist in efr log, but still no Sigma msg come back, is there something wrong?
> Updating device address to UDP:[fd3d:776c:a6df:29fc:862e:5a77:6c7c:e6e5]:5540 while in state 1

in efr log:
> <detail> [DL]    Thread Unicast Addresses:[2022-06-20 10:26:49.448]
> <detail> [DL]         fd3d:776c:a6df:29fc:862e:5a77:6c7c:e6e5/64 valid preferred[2022-06-20 10:26:49.448]
> <detail> [DL]         fd7f:a92c:bbb8:293:0:ff:fe00:4c02/64 valid rloc[2022-06-20 10:26:49.448]
> <detail> [DL]         fd7f:a92c:bbb8:293:1c25:72f5:cd4d:f122/64 valid[2022-06-20 10:26:49.448]
> <detail> [DL]         fe80:0:0:0:1096:7bea:3229:5ff5/64 valid preferred[2022-06-20 10:26:49.448]

[[efr log]matter light-switch demo pair failed.txt](https://github.com/project-chip/connectedhomeip/files/8937105/efr.log.matter.light-switch.demo.pair.failed.txt)
[[chiptool log]matter light-switch demo pair failed.txt](https://github.com/project-chip/connectedhomeip/files/8937106/chiptool.log.matter.light-switch.demo.pair.failed.txt)

bzbarsky-apple, 2022-06-22:
> and I tried remove /tmr/chip*, it's suppose to clean all of chip cache to update the unicast address.

That won't remove system DNS caches....

Looking at [[chiptool log]matter light-switch demo pair failed.txt](https://github.com/project-chip/connectedhomeip/files/8937106/chiptool.log.matter.light-switch.demo.pair.failed.txt) it looks like there's an attempt to commission as node id 1234.  

This, as you note, discovers `fd3d:776c:a6df:29fc:862e:5a77:6c7c:e6e5` as the IP and sends Sigma1 to it.  And this does seem to be the address of the device.  But te device never receives the Sigma1.

Was that IP address ping-able from the machine chip-tool was running on?  This sounds like a networking configuration issue...

Leo2442926161, 2022-06-22:
is it possible that I add the --sed in build parameter? it can pair easily  after I removed --sed parameter.

which commit is cfc35951be66a664a6efdadea56d1b8ea6e63e96

bzbarsky-apple, 2022-06-22:
> it can pair easily after I removed --sed parameter.

Interesting.  @jmartinez-silabs could you take a look, please?

jmartinez-silabs, 2022-06-23:
I'll take a look into it

jmartinez-silabs, 2022-07-04:
Can you test --sed again with this PR from Mathieu that I just merge 
https://github.com/project-chip/connectedhomeip/pull/20184

It should fix the issue. We should also have another soon that reduces significantly the power consumption

closed 2023-01-08 by stale[bot]