project-chip/connectedhomeip #19676: [TC-DD-3.17] - DUT should terminate the commissioning process when we pass invalid QR Code, but DUT successfully completed the commissioning process

sowmyassp, 2022-06-16:
**Issue** : According to test-plan of [TC-DD-3.17] In Step 6b its mentioned DUT should terminate the commissioning process when we pass invalid 21 digit manual code generated in step 6a but DUT successfully completed the commissioning process.

**Expected behaviour** : DUT should terminate the commissioning mode when invalid 21 digit manual code is passed

**Actual behaviour** : DUT successfully completed the commissioning process.

**Commit used** : c1e1d018820c0fdd51f124f607b9207f31fae2ff			

**Steps to reproduce** :
**DUT side** :
sudo rm -rf /tmp/chip_*
sudo ./chip-all-clusters-app --wifi (Discriminator - 3841, Passcode - 20202021)

**TH side**
1. ./out/chip-tool/chip-tool pairing manualcode 1 749701123365521327694(0xFFF1)
2. ./out/chip-tool/chip-tool pairing manualcode 1 749701123365522327692(0xFFF2)
3.  ./out/chip-tool/chip-tool pairing manualcode 1 749701123365523327697(0xFFF3)
4. ./out/chip-tool/chip-tool pairing manualcode 1 749701123365524327693(0xFFF4)

**App used** - all-clusters-app

**Test platform** :
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM

**Test-plan reference** :
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicediscovery.adoc#tc-dd-3-17-commissioning-flow-21-digit-manual-pairing-code-negative-scenario-dut-commissioner

**PFA Logs:**

[TC_DD_3.17_dut.txt](https://github.com/project-chip/connectedhomeip/files/8918788/TC_DD_3.17_dut.txt)
[TC_DD_3.17_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8918789/TC_DD_3.17_cntrl.txt)

bzbarsky-apple, 2022-06-16:
What the test plan says here is:

> DUT terminates the commissioning process in a DUT-specific manner according to the DUT manufacturer’s instructions, ***unless the user is made fully aware of the security risks of providing an uncertified device with operational and networking credentials***. [emphasis mine]

The test plan does NOT require a DUT to fail to commission these vendor ids; it just requires that the user be clearly notified that these are uncertified devices.

And even that notification is not reasonable as a _default_ behavior for our test tools, because commissioning things with test vendor ids is by far the most common use of those test tools.

I think we need to figure out exactly what this test plan means in practice for _our test tools_ and whether it applies to them at all.  For example, does using a test tool constitute a priori notification that there are risks?

andy31415, 2022-06-17:
```
./out/linux-x64-chip-tool/chip-tool payload parse-setup-payload 749701123365521327694
[1655479375.139081][3332995:3332995] CHIP:SPL: Parsing decimalRepresentation: 749701123365521327694
[1655479375.139101][3332995:3332995] CHIP:SPL: Version:       0
[1655479375.139103][3332995:3332995] CHIP:SPL: VendorID:      65521
[1655479375.139104][3332995:3332995] CHIP:SPL: ProductID:     32769
[1655479375.139105][3332995:3332995] CHIP:SPL: Custom flow:   2    (CUSTOM)
[1655479375.139106][3332995:3332995] CHIP:SPL: Capabilities:  0x00 (NONE)
[1655479375.139107][3332995:3332995] CHIP:SPL: Discriminator: 3840
[1655479375.139108][3332995:3332995] CHIP:SPL: Passcode:      20202021
```

is the Custom flow an issue?

andy31415, 2022-06-17:
Reading more comments, I guess the request is 'show a nice message about this being a test id' .... It sounds like that is a nice to have (guessing not a spec SHALL requirement) and probably not a cert blocker for our internal test tools like chip_tool.

You can make a test plan to use `parse-setup-payload` during certification to validate ids that vendors generate. But that seems independent of tooling or SDK examples.

andy31415, 2022-06-17:
So it sounds like this is a test case for "commissioners". We have to require any commissioner to show proper notices.

Does this extend to chip-tool though? I guess it could and we can add some text. I also believe that chip-tool is a wall-of-text in terms of the output so any warnings will be missed. To be in the spirit of the spec, I believe we should reject by default and have some flag of `--allow-test-ids` so that the user has to manually allow them ... but that will get annoying fast for developers.

andy31415, 2022-06-17:
Are we testing DUT or testing chip-tool as a commissioner. I believe former, so then this is not a spec blocker for the SDK.

andy31415, 2022-06-17:
To expand on this: `cert blocker` is if you cannot prove that you can write a spec-compliant app using the SDK.
SDK proves it can parse qr codes and display them (using `payload parse-setup-payload`) so an implementation can display any necessary texts.

Unless we want to certify chip-tool, I don't think there is an issue here.

PR #19735 (2022-07-06): Add payload parser to darwin framework tool
#### Problem
Darwin-framework-tool does not have a payload parser to demonstrate that a commissioner implemented using CHIP.framework can indeed parse out the Vendor ID and check to see if it is valid. 
* Fixes #19676

#### Change overview
- Add a parser to Darwin-framework-tool

#### Testing
- Parsed SetupQRCode
- Parsed ManualCode