project-chip/connectedhomeip #19678: Closing the commissioning window should not expire fail-safe

tehampson, 2022-06-16:
#### Problem
With Max Cumulative Fail safe timer created we now should no longer expire the failsafe timer when the commission window closes either due to open window timeout or RevokeCommissioning command

tehampson, 2022-06-16:
Does [the spec need to be updated](https://github.com/CHIP-Specifications/connectedhomeip-spec/blame/master/src/service_device_management/CommissioningFlows.adoc#L75). In section 5.5 I see:
```
A Commissionee that is ready to be commissioned SHALL accept the request to establish a PASE session with the first
Commissioner that initiates the request. When a Commissioner is either in the process of establishing a PASE session
with the Commissionee or has successfully established a session, the Commissionee SHALL NOT accept any more
requests for new PASE sessions until session establishment fails or the successfully established PASE session is
terminated on the commissioning channel (see CloseSession in Secure Channel Status Report Messages). In the
event a CloseSession status message is sent or received:

1. If the fail-safe timer is armed, the fail-safe timer SHALL be considered expired and the cleanup steps detailed
in Section 11.9.7.2, “ArmFailSafe Command” SHALL be executed.

2. If the commissioning window is still open, the Commissionee SHALL continue listening for commissioning requests.
```

mrjerryjohns, 2022-06-16:
^^ @tcarmelveilleux @bzbarsky-apple

bzbarsky-apple, 2022-06-16:
What is the spec update being proposed?

What the spec says right now is:

1. When the commissioning window opens, start listening for incoming PASE.
2. If a PASE session is established, arm a fail-safe.
3. If the fail-safe expires, close the PASE session.
4. If the PASE session is closed, expire the fail-safe.  Note that this means that the lifetime of the fail-safe exactly matches the lifetime of the PASE session.
5. If the fail-safe expires (aka PASE session is closed), if the commissioning window is still open go back to listening for incoming PASE.

bzbarsky-apple, 2022-06-16:
And as of yesterday I believe the SDK implements all of those bits, by the way.

tehampson, 2022-06-16:
I don't have a proposed update, more so I am trying to understand the discussion around https://github.com/project-chip/connectedhomeip/pull/19636/files#r898314839 and how it relates to making sure we do not expire the fail-safe timer when the commission window closes either due to open window timeout or `RevokeCommissioning` command.

I was under the impression that is the open window times out, or `RevokeCommissioning` is called that and current PASE session would then be closed, which (based on spec today) means that we expire the fail-safe. So either I am missing something or that `ExpireFailSafeIfArmed();` in `CommissioningWindowManager::Cleanup()` is actually reasonable

tehampson, 2022-06-16:
I see where I have a misunderstanding:

The `CommissioningTimeout` applies only to cessation of any announcements and to accepting of new commissioning sessions; it does not apply to abortion of connections, i.e., a commissioning session SHOULD NOT abort prematurely upon expiration of this timeout.

So closing of the window doesn't terminate the PASE connection

bzbarsky-apple, 2022-06-16:
> So closing of the window doesn't terminate the PASE connection.

Right, this is very important.  :)

closed 2022-06-22 by tehampson