project-chip/connectedhomeip #19680: [TC-DD-3.14] - DUT should terminate the commissioning process when we pass invalid QR Code, but DUT successfully completed the commissioning process

sowmyassp, 2022-06-16:
**Issue** : According to test-plan of [TC-DD-3.14] In Step 3b its mentioned DUT should terminate the commissioning process when we pass invalid QR code generated in step 3a but DUT successfully completed the commissioning process.
Similarly the commissioning process is completed for step 3c, 4a, 4c when we pass invald QR code

**Expected behaviour** : DUT should terminate the commissioning mode when invalid QR code is passed

**Actual behaviour** : DUT successfully completed the commissioning process.

**Commit used** : c1e1d018820c0fdd51f124f607b9207f31fae2ff			

**App used** - all-clusters-app

**Steps to reproduce** :
**DUT side** :
sudo rm -rf /tmp/chip_*
sudo ./chip-all-clusters-app --wifi (Discriminator - 3841, Passcode - 20202021)
**TH side**
1. For Custom Flow field, set it to 1: User-Intent Commissioning Flow - invalid QR code is "MT:-24J06VO00KA0648G00"
    /out/chip-tool/chip-tool pairing qrcode 1 MT:-24J06VO00KA0648G00
2. For Custom Flow field, set it to 2: Custom Commissioning Flow - invalid QR code is "MT:-24J029Q00KA0648G00"
    /out/chip-tool/chip-tool pairing qrcode 1 MT:-24J029Q00KA0648G00
3. For Discovery Capability bit string, set it to BLE for discovery - invalid QR code is "MT:-24J0YXE00KA0648G00"
    ./out/chip-tool/chip-tool pairing code 1 MT:-24J0YXE00KA0648G00
4. For  Discovery Capability bit string, set it to IP Network for discovery - invalid QR code is "MT:-24J029Q00KA0648G00"
   ./out/chip-tool/chip-tool pairing code 1 MT:-24J029Q00KA0648G00

**Test platform** :
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM

**Test-plan reference** :
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicediscovery.adoc#tc-dd-3-14-commissioning-flow-qr-code-negative-scenario-dut-commissioner

**PFA Logs**:
[TC_DD_3.14_dut_CFL_2.txt](https://github.com/project-chip/connectedhomeip/files/8919289/TC_DD_3.14_dut_CFL_2.txt)
[TC_DD_3.14_dut_CFL_1.txt](https://github.com/project-chip/connectedhomeip/files/8919290/TC_DD_3.14_dut_CFL_1.txt)
[TC_DD_3.14_dut_capability_2.txt](https://github.com/project-chip/connectedhomeip/files/8919291/TC_DD_3.14_dut_capability_2.txt)
[TC_DD_3.14_dut_capability_1.txt](https://github.com/project-chip/connectedhomeip/files/8919292/TC_DD_3.14_dut_capability_1.txt)
[TC_DD_3.14_cntrl_CFL_2.txt](https://github.com/project-chip/connectedhomeip/files/8919293/TC_DD_3.14_cntrl_CFL_2.txt)
[TC_DD_3.14_cntrl_CFL_1.txt](https://github.com/project-chip/connectedhomeip/files/8919294/TC_DD_3.14_cntrl_CFL_1.txt)
[TC_DD_3.14_cntrl_capability_2.txt](https://github.com/project-chip/connectedhomeip/files/8919295/TC_DD_3.14_cntrl_capability_2.txt)
[TC_DD_3.14_cntrl_capability_1.txt](https://github.com/project-chip/connectedhomeip/files/8919296/TC_DD_3.14_cntrl_capability_1.txt)

bzbarsky-apple, 2022-06-16:
This is lumping together completely different things, which need different actions/ resolutions.  @sowmyassp Please file one issue per issue.  There should be 4 separate issues here, each with a clear description of the expected behavior _and spec references to back up that clear description_.

sowmyassp, 2022-06-16:
Thank you @bzbarsky-apple

closed 2022-06-17 by sowmyassp