project-chip/connectedhomeip #19683: Chip-tool commissioning lighting app - Failed to perform commissioning step 18

rwells1703, 2022-06-16:
Hello, I'm having an issue on a Linux x64 build of chip-tool. Currently I'm trying to get chip-tool to commission the example lighting-app over ethernet, but it fails on step 18. The chip tool is running on a Ubuntu 20.04 x64 PC and the lighting-app is running on a Debian 5.10.84-1 x64 PC.

Not sure if its possibly related to #18753, as it fails at the same step. Both devices are on the same local network, connected via Ethernet and there is no firewall between them (I've also check iptables/ip6tables for any rules blocking traffic, but there are none).

I have noticed a few ```CHIP:DIS: DNSSD parse error: ../../third_party/connectedhomeip/src/lib/dnssd/IncrementalResolve.cpp:322: CHIP Error 0x0000000B: No memory``` errors in the logs, however I think they are to do with #18941 instead.

Has anyone experienced anything similar, any help would be greatly appreciated!

## Attempt 1: Letting chip-tool automatically discover the address (commissioning fails)
I've tried a number of different ways to initiate commissioning, for example letting chip-tool discover the device automatically and using:
```$ sudo ./chip-tool pairing onnetwork 1234 20202021```

And lighting-app with the command:
```$ sudo ./chip-lighting-app```

Here are the logs:
[chip-tool-onnetwork-log.txt](https://github.com/project-chip/connectedhomeip/files/8919789/chip-tool-onnetwork-log.txt)
[lighting-app-onnetwork-log.txt](https://github.com/project-chip/connectedhomeip/files/8919791/lighting-app-onnetwork-log.txt)

Using this method, I suspect that chip-tool is trying to connect to the lighting-app using one of its advertised IPv6 addresses, however this seems to fail for some reason. The test bulb device appears when you run ```$ sudo ./chip-tool discover commissionables``` so I don't think the issue is the bulb not being found:

[image]

## Attempt 2: Specifying the IPv4 address and device port manually (commissioning fails at step 18)
I ran chip-tool with the command:
```$ sudo ./chip-tool pairing ethernet 1234 20202021 3840 192.168.202.94 5540```

And lighting-app with the command:
```$ sudo ./chip-lighting-app```

Here are the logfiles:
[chip-tool-log.txt](https://github.com/project-chip/connectedhomeip/files/8919752/chip-tool-log.txt)
[lighting-app-log.txt](https://github.com/project-chip/connectedhomeip/files/8919754/lighting-app-log.txt)

From what I can see in Wireshark, communication seems to change over to IPv6 for some reason (see screenshot):

[image]

## Attempt 3: Running chip-tool and lighting-app on the same PC (works)
I've also tried running both chip-tool and lighting-app on the same PC, which works and commissioning functions as expected, even letting chip-tool discover the device automatically with:
```$ sudo ./chip-tool pairing onnetwork 1234 20202021```

And lighting-app with the command:
```$ sudo ./chip-lighting-app```

Here are the logs showing it functioning correctly:
[chip-tool-same-pc-log.txt](https://github.com/project-chip/connectedhomeip/files/8919556/chip-tool-same-pc-log.txt)
[lighting-app-same-pc-log.txt](https://github.com/project-chip/connectedhomeip/files/8919504/lighting-app-same-pc-log.txt)

For comparison with above, here is a Wireshark packet capture over the loopback interface:

[image]

rwells1703, 2022-06-17:
Hi thanks for your response - so when I submitted the issue I was on f3247ba900c4d372176d239142463970396c4888. I tried it again yesterday, after recompiling on 3963c1b1a36d6e5914c3471a4fd48949b50417ed and the same issue persisted. It is still getting stuck after chip-tool sends the Sigma1 message. 

I've tried pinging the IP addresses advertised on discover and they all respond **apart from** the link local address ```fe80::225:64ff:fea7:fceb```, which gives:

[image]

This is the ifconfig for the machine running lighting-app:

[image]

Is it likely that chip-tool is sending its trying to initiate the CASE handshake over this link-local IP, instead of the others, hence why there is no response?

rwells1703, 2022-06-17:
Just to clarify, the IP addresses have changed since yesterday. This is currently what the lighting-app advertises:

[image]

Curiously it appears identically 4 times in the log when running ```sudo chip-tool discover commissionables```:

[discover.txt](https://github.com/project-chip/connectedhomeip/files/8926379/discover.txt)

rwells1703, 2022-06-17:
Okay my mistake, it seems I can ping the link local address ```fe80::225:64ff:fea7:fceb```. I just had to provide the network interface in the address like so:

[image]

So unfortunately I'm still not sure why the lighting-app wont respond. Do you have any more suggestions for me to try? Thanks!

rwells1703, 2022-06-17:
Hmm I've found a fix for the issue on 3963c1b1a36d6e5914c3471a4fd48949b50417ed, its a very strange fix though. Fairly sure its related to my internal network routing like you mentioned.

Oddly the fix I discovered is to constantly ping all the IP addresses belonging to the device running chip-tool, from the device running lighting-app. I discovered that while trying to connect to IPv6 servers over the local network e.g. SCP (uses SSH) and a web server, the servers would not respond unless first pinged. You can then connect to them and they work as usual for a short period of time (maybe ~30s) before they stop working and need to be pinged to work again. The same applies to here I think. My first thought was it was a firewall, however I have none setup on either device - nor do I have any port knocking enabled.

Interesting to note that pairing does not work on the latest commit 5ea0ca66be445a5a4ac771c483412c10fba022e7, where it fails with an internal error, whereas it works fine on the 3963c1b1a36d6e5914c3471a4fd48949b50417ed build from yesterday.

Would like to thank you for analysing the logs though, was very helpful!

bzbarsky-apple, 2022-06-17:
> Is it likely that chip-tool is sending its trying to initiate the CASE handshake over this link-local IP

Fwiw, no.  Per logs it was using `2a00:23a8:89b:da01:45b2:6372:b224:91e5`.

I don't know what could be leading to the behavior you observe where pings affect ipv6 connectivity...  I'll reach out to some people who might know more there.

bzbarsky-apple, 2022-06-17:
I suspect people will want to know more information about the network (e.g. which router is being used), by the way.

turon, 2022-06-18:
Yes, @rwells1703 could you provide more info on the setup?

What AP are you using (brand model #).

Are both client and server Linux nodes on WiFi or is there a Thread hop / border router involved too?

Are the Linux nodes laptops or RasPis?

Thanks.  If you could provide the full topology and list of hardware used it may help isolate the issue.

High-level, it seems your AP may be filtering LL UDP traffic.

closed 2022-12-30 by stale[bot]