project-chip/connectedhomeip #19706: ACL behavior depends on link order because of how privilege-storage.cpp and RequiredPrivilege.cpp interact

bzbarsky-apple, 2022-06-17:
#### Problem
`src/app/RequiredPrivilege.cpp` and `src/app/util/privilege-storage.cpp` both define functions with the same names; the former defines them as weak symbols, while the latter defines them as non-weak symbols.

Unfortunately, _all_ the symbols in `privilege-storage.cpp` match names in `RequiredPrivilege.cpp`, so if the static library containing `RequiredPrivilege.cpp` happens to come earlier than the static library containing `privilege-storage.cpp` while linking, we will not end up pulling in the `privilege-storage.cpp` symbols, since those names will all be satisfied already.

The failure mode here is pretty insiduous: the application compiles/links correctly, but all ACL checks require "administer" privilege... which means the problem would not necessarily be caught in any of our normal testing, if people are generally testing with commissioners that do in fact have "administer" privileges.

 #### Proposed Solution
Either stop using weak symbols or add some other function to `privilege-storage.cpp` that is called from somewhere and forces those symbols to be linked in.  Any better options?

andy31415, 2022-06-17:
Could we devise a test to catch this issue (i.e. intentionally trigger it then verify our fix like the linking bit?).

bzbarsky-apple, 2022-06-17:
Do you mean a CI test or a certification test?

msandstedt, 2022-08-31:
Is this resolved by https://github.com/project-chip/connectedhomeip/pull/21220?

bzbarsky-apple, 2022-08-31:
Yes, it is!

closed 2022-08-31 by bzbarsky-apple