project-chip/connectedhomeip #19709: [Pre-SVE] access-control-entry-changed is emitted on maxInterval Time even if event was added earlier

doublemis1, 2022-06-17:
#### Problem
While subscribing to Access Control event access-control-entry-changed (valid to check for the second event), the information about the change of the acls is not on change time but on max interval time.

Steps:
```
commission device
./chip-tool interactive start
accesscontrol subscribe-event access-control-entry-changed 1 100 1 0
accesscontrol write acl [{"fabricIndex":1,"privilege":5,"authMode":2,"subjects":[112233],"targets":[{"cluster":null,"endpoint":0,"deviceType":null}]},{"fabricIndex":1,"privilege":3,"authMode":2,"subjects":null,"targets":[{"cluster":null,"endpoint":1,"deviceType":null}]}] 1 0
```

[acl.log](https://github.com/project-chip/connectedhomeip/files/8926070/acl.log)

 #### Proposed Solution
Events should be emitted at the change time.

bzbarsky-apple, 2022-06-17:
The event is emitted (in the sense of being added to the event buffer) at change time.

The real issue is that we don't seem to flag `ReadHandler` instances as "has something to report" due to events being queued, only due to attributes changing.  As a result, events are reported either "immediately", ignoring min-interval, if they are "urgent", or reported at the max interval...  That definitely seems like a problem.

@mrjerryjohns @yunhanw-google why is the behavior what it is right now?

bzbarsky-apple, 2022-06-17:
Compare to what it says for attributes:

> Attribute changes SHALL be delivered as soon as possible, taking into account the minimum interval.

which means events and attributes should have the same behavior.

doublemis1, 2022-06-17:
Quick information, as far as I remember the state-transition event from otasoftwareupdaterequestor cluster works as you described. The Event was sent in a range of min and max intervals while the event was added to the buffer.

mrjerryjohns, 2022-07-20:
There is a linked [spec](https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5383) issue that @robszewczyk is going to help tackle to clarify this.

mrjerryjohns, 2022-07-20:
If the wording shakes out the way I think it will based on chats with @robszewczyk, then there should be no extra work for the SDK to do.

woody-apple, 2022-07-27:
Let's close this then.

bzbarsky-apple, 2022-08-02:
> If the wording shakes out the way I think it will based on chats with @robszewczyk, then there should be no extra work for the SDK to do.

I don't think that's true.  The wording being proposed in the spec does _not_ seem to be what the SDK implements.  Reopening until we get this sorted.

woody-apple, 2022-08-11:
Review with @bzbarsky-apple. Sounds like @yunhanw-google will be looking at this. @mrjerryjohns ?

mrjerryjohns, 2022-08-15:
Given the most recent spec changes that permit the publisher to queue up non-urgent events the only thing that needs to happen now is to fix the logic in the reporting engine to ensure eventing honors the min-interval (which we already have a dedicated issue for: #21864).

From the perspective of this issue, there is nothing to be done.

closed 2022-08-15 by mrjerryjohns