project-chip/connectedhomeip #19712: [Pre-SVE] TC-G-2.2 step 12b value mismatch

AlvinHsiao, 2022-06-17:
Problem:
The group-table value of DUT is different from the CHIP tool Verification steps.

Expected behavior: The group-table value of DUT should be identical to the CHIP tool Verification steps.
Actual behavior: The group-table value of DUT is different from the CHIP tool Verification steps.
steps to reproduce:-./chip-tool groupkeymanagement read group-table 33 0

system configuration
Chip-Tool: RPI
DUT used : Infineon CYW30739
app use: lighting-app
chip-tool and DUT commit id : https://github.com/project-chip/connectedhomeip/commit/5d8599d195a93dfe5473f64e4f888c322e5df1d2 (pre-sve SHA)

logs
[TC-G-2.2.txt](https://github.com/project-chip/connectedhomeip/files/8926467/TC-G-2.2.txt)

bzbarsky-apple, 2022-06-17:
@AlvinHsiao can you please point me to the verification steps in question?  That is, what value was expected, and how did it differ from the value that was found?

bzbarsky-apple, 2022-06-17:
Oh, nevermind, this is about RemoveAllGroups.

bzbarsky-apple, 2022-06-17:
Duplicate of #19610