project-chip/connectedhomeip #19720: BTP handshake message version array is not spec compliant (order flipped)

andy31415, 2022-06-17:
#### Problem
Spec says versions are "high nibble first" according to tables.  SDK uses low nibble first.

[image]
[image]
[image]

 #### Proposed Solution
Fix SDK. Also remove early exit on 'list end' to remain compatible with non-spec compliant old code.

andy31415, 2022-06-17:
If https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5373 lands, this becomes a non issue (spec flips the order of the array). 

Either this issue has to be fixed in the SDK or spec issue has to be fixed in spec. Not both because then we flip-flop ...

woody-apple, 2022-06-21:
Cert Blocker Review: Making this a spec conformance issue, as there's no cert test for this that we know of.

andy31415, 2022-07-04:
https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5373 is ready to be merged - this issue should be closed once that PR is merged.