project-chip/connectedhomeip #19726: The YAML "contains" does not seem to work when used to validate AttributeList

manjunath-grl, 2022-06-17:
Summary:
When we tried to use the "contains" [https://github.com/project-chip/connectedhomeip/pull/19042] keyword to validate the mandatory attribute-ids the YAML script build seems to fail, indicating a type mismatch. We refereed the samples provided in the "TestConstraints.yaml" when trying this out, are we missing anything with respect to the usage of this keyword ?

YAML script:
[Test_TC_MC_1_3.zip](https://github.com/project-chip/connectedhomeip/files/8929384/Test_TC_MC_1_3.zip)
Execution logs for reference:
[image]
[Contains_keyword error.zip](https://github.com/project-chip/connectedhomeip/files/8929402/Contains_keyword.error.zip)

bzbarsky-apple, 2022-06-17:
@manjunath-grl I'm going to ask for the last time: when filing issues like this, please point to a diff that can be used to reproduce the problem.  Attaching just a single yaml file is not ok.  Especially because in this case that single file does not codegen properly if placed in the right spot in the tree.

@naveenkommareddi

manjunath-grl, 2022-06-17:
@bzbarsky-apple sorry about that, I see a PR that fixes the issues is already linked, we will make sure to include the diff moving forward. linking the changes [here](https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:Issue_19726) just in case its needed.