project-chip/connectedhomeip #19727: [pre-sve]TC-CNET-4.4 Bad response

chrisbeach59, 2022-06-17:
Expected a list of networks got:
[1655486825.864859][41820:41825] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0001
[1655486825.864954][41820:41825] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0001
[1655486825.865146][41820:41825] CHIP:TOO:   ScanNetworksResponse: {
[1655486825.865242][41820:41825] CHIP:TOO:     networkingStatus: 0
[1655486825.865359][41820:41825] CHIP:TOO:     wiFiScanResults: 7 entries
[1655486825.865439][41820:41825] CHIP:TOO:       [1]: {
[1655486825.865489][41820:41825] CHIP:TOO:         Security: 0
[1655486825.866616][41820:41825] CHIP:TOO:          Struct truncated due to invalid value for 'Ssid'
[1655486825.866680][41820:41825] CHIP:TOO:    }

bzbarsky-apple, 2022-06-17:
@chrisbeach59 What server app is running?  On what platform?  What SHAs are being used?  What was the actual payload (logged just above the part you're quoting)?  Please attach (not paste) the full log.

chrisbeach59, 2022-06-19:
SHA 1e3f12041013dd71a53d035dbf2f6d142fc06c4e
I am running my thermostaat app on the Ameba/Resideo pllatform 
Log file attached
[TC-CNET-4-4.log](https://github.com/project-chip/connectedhomeip/files/8935725/TC-CNET-4-4.log)

bzbarsky-apple, 2022-06-21:
Thanks, that is very helpful.  So the TLV dump shows the SSID values as empty byte strings.   `TLVReader::Get(ByteSpan &)` will return a ByteSpan whose `data()` is null and length is 0 in that situatuon.

The chip-tool code for logging a byte string does `BytesToUppercaseHexString` which calls `BytesToHex`.  `BytesToHex` does:
```
    if ((src_bytes == nullptr) || (dest_hex == nullptr))
    {
        return CHIP_ERROR_INVALID_ARGUMENT;
    }
```
which will fail out if `src_bytes` is null, even if `src_size == 0`.

We could fix this either in chip-tool (by not calling `BytesToHex` with the empty buffer), or in `BytesToHex`, by having it handle null + 0-length correctly....

bzbarsky-apple, 2022-06-21:
I guess `BytesToHex` does document that the buffer must not be null...  @tcarmelveilleux would you prefer we allow null + 0-size here, or just fix this caller?

tcarmelveilleux, 2022-06-21:
Let's handle null + 0-length correctly

woody-apple, 2022-06-21:
Cert Blocker Review: Assigning to @bzbarsky-apple

PR #19763 (2022-06-22): [Ameba] Use BLE random address and add uxTaskGetFreeStackSize & uxTaskGetStackSize
#### Problem
- Config BLE using random address
- Add uxTaskGetFreeStackSize & uxTaskGetStackSize
- Need to exclude SSID with length 0 during scan-network

#### Change overview
* Update chip-build-ameba/Dockerfile
* Update version
* Fixes #19727

#### Testing
Tested docker build
closed 2022-06-22 by tcarmelveilleux