project-chip/connectedhomeip #19729: [pre-sve]TC-CNET4.4 Step 2 bad response

chrisbeach59, 2022-06-17:
./chip-tool networkcommissioning scan-networks 1111 0 --Ssid 574E38303254 --Breadcrumb 1
the Sdid  specified matches the current network  read from
./chip-tool networkcommissioning read networks 1111 0

[1655488863.152775][41969:41974] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0001
[1655488863.152900][41969:41974] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0001
[1655488863.153041][41969:41974] CHIP:TOO:   ScanNetworksResponse: {
[1655488863.153157][41969:41974] CHIP:TOO:     networkingStatus: 0
[1655488863.153252][41969:41974] CHIP:TOO:     wiFiScanResults: 0 entries

I would expect to see the network information for the connected network.

chrisbeach59, 2022-06-19:
this relates to https://github.com/project-chip/connectedhomeip/issues/19729

chrisbeach59, 2022-06-19:
[TC-CNET-4-4-2.log](https://github.com/project-chip/connectedhomeip/files/8935730/TC-CNET-4-4-2.log)

pankore, 2022-06-20:
@chrisbeach59, according to the test step, Step 2: "TH sends ScanNetworks Command to the DUT with SSID argument value as 'Userwifi_ssid' and Breadcrumb argument value 1"
I think this means you should pass in the ASCII value of ssid instead of hex. I have tested it with Ameba platform and got the expected result.
Yet, in the example verification step, the hex value is passed in, might be a mistake?

closed 2022-12-30 by stale[bot]