project-chip/connectedhomeip #19755: [pre-sve] TC-CNET-4.11 chip-tool timeout before ConnectMaxTimeSeconds

pankore, 2022-06-20:
Platform: Ameba all-clusters-app
SHA: 1e3f12041013dd71a53d035dbf2f6d142fc06c4e

 #### Problem
- Step 5, when connect-network command is issued, chip-tool timeout before ConnectMaxTimeSeconds. Ameba is not able to connect to AP and obtain IPv6 address in time.
- ConnectMaxTimeSeconds was 20seconds.

 #### Proposed Solution
Fix chip-tool to wait for ConnectMaxTimeSeconds

bzbarsky-apple, 2022-06-20:
@pankore Which _exact_ chip-tool commands were run in this case?  If you're running the "pairing" commands they should be waiting long enough.  If you're manually running the ConnectNetwork manually, then chip-tool is not going to try to second-guess what you want it to do and will use the default command timeout.  If you want a different timeout, you can specify that via the `--timeout` argument to the command.

woody-apple, 2022-06-21:
Cert Blocker Review: @pankore Can you please add logs, there's no information here to help debug this. Thanks!

closed 2022-06-22 by pankore