project-chip/connectedhomeip #19758: TC-MF-1.22: can only commission N-1 fabrics

sstruchtrup, 2022-06-20:
#### Problem

Observed on our DUT, but reproduced with the chip-lighting-app on Linux, with 16 supported fabrics, only 15 fabrics can be commissioned.

DUT fails with CHIP Error 0x0000000B: No memory:
```
[1655709316.435309][8634:8634] CHIP:EM: Received message of type 0x30 with protocolId (0, 0) and MessageCounter:76643446 on exchange 49310r
[1655709316.435427][8634:8634] CHIP:EM: Handling via exchange: 49310r, Delegate: 0x55f3e9195fb0
[1655709316.435478][8634:8634] CHIP:IN: CASE Server received Sigma1 message. Starting handshake. EC 0x55f3e9196598
[1655709316.435506][8634:8634] CHIP:IN: CASE Server enabling CASE session setups
[1655709316.435522][8634:8634] CHIP:EM: Sending Standalone Ack for MessageCounter:76643446 on exchange 49310r
[1655709316.435551][8634:8634] CHIP:IN: Prepared unauthenticated message 0x7ffc13eba610 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 49310r with MessageCounter:1436599020.
[1655709316.435610][8634:8634] CHIP:IN: Sending unauthenticated msg 0x7ffc13eba610 with MessageCounter:1436599020 to 0x0000000000000000 at monotonic time: 00000000002EA7EB msec
[1655709316.435767][8634:8634] CHIP:EM: Flushed pending ack for MessageCounter:76643446 on exchange 49310r
[1655709316.435831][8634:8634] CHIP:EM: OnMessageReceived failed, err = ../../third_party/connectedhomeip/src/protocols/secure_channel/PairingSession.cpp:29: CHIP Error 0x0000000B: No memory
```

chip-tool fails with:
```
[1655709371.409802][27697:27698] CHIP:SC: CASESession timed out while waiting for a response from the peer. Current state was 1
[1655709371.409877][27697:27698] CHIP:IN: SecureSession Released 0x5626dfaeb928 Type:2 LSID:33303
[1655709371.409901][27697:27698] CHIP:CTL: OperationalDeviceProxy[A52AD70C0D78A409:0000000000000001]: State change 4 --> 3
[1655709371.409952][27697:27698] CHIP:CTL: Device connection failed. Error ../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:246: CHIP Error 0x00000032: Timeout
[1655709371.409979][27697:27698] CHIP:CTL: Error on commissioning step 'FindOperational': '../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:246: CHIP Error 0x00000032: Timeout'
[1655709371.409993][27697:27698] CHIP:CTL: Failed to perform commissioning step 18
```

This looks similar to #19642, but the error messages are different.

sstruchtrup, 2022-06-20:
Full DUT log (chip-lighting-app): [log-DUT-19758.txt](https://github.com/project-chip/connectedhomeip/files/8938399/log-DUT-19758.txt)

chip-tool output for the commissioning of the 16th fabric: [chip-tool-16th-commissioning.txt](https://github.com/project-chip/connectedhomeip/files/8938413/chip-tool-16th-commissioning.txt)

sstruchtrup, 2022-06-20:
Tested with the pre-SVE commit 425f1fb916ca154deeb75cd82cd5196778aba152.

chip-lighting-app for linux configured as follows:
```shell
gn gen out/host --args="target_os=\"linux\"
        chip_inet_config_enable_ipv4=false
        chip_enable_openthread=false
        chip_enable_ota_requestor=false
        chip_config_network_layer_ble=false
        chip_mdns=\"minimal\"
        chip_enable_ble=false
        chip_enable_wifi=false
        treat_warnings_as_errors=false"
```

with the following patch:
```diff
diff --git a/src/platform/Linux/BUILD.gn b/src/platform/Linux/BUILD.gn
index bf40d4b84..27be21aba 100644
--- a/src/platform/Linux/BUILD.gn
+++ b/src/platform/Linux/BUILD.gn
@@ -63,8 +63,6 @@ static_library("Linux") {
     "Logging.cpp",
     "NetworkCommissioningDriver.h",
     "NetworkCommissioningEthernetDriver.cpp",
-    "NetworkCommissioningThreadDriver.cpp",
-    "NetworkCommissioningWiFiDriver.cpp",
     "PlatformManagerImpl.cpp",
     "PlatformManagerImpl.h",
     "PosixConfig.cpp",
```

sstruchtrup, 2022-06-20:
```
# chip-tool operationalcredentials read supported-fabrics 1 0
[1655710403.313292][27924:27926] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_003E Attribute 0x0000_0002 DataVersion: 1164008626
[1655710403.313357][27924:27926] CHIP:TOO:   SupportedFabrics: 16
```

```
# chip-tool operationalcredentials read fabrics 1 0 --fabric-filtered 0
[1655710441.935578][27927:27928] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_003E Attribute 0x0000_0001 DataVersion: 1164008626
[1655710441.935686][27927:27928] CHIP:TOO:   Fabrics: 15 entries
[1655710441.935716][27927:27928] CHIP:TOO:     [1]: {
[1655710441.935733][27927:27928] CHIP:TOO:       RootPublicKey: 040535C04FF8D9105DBAFF9F940DE6D04CEA09F292A354A32E92636771C65568BE896F454E9B524FD1797C399AD388335574855806
[1655710441.935760][27927:27928] CHIP:TOO:       .............: AE1B434D3809C2F07937831E
[1655710441.935773][27927:27928] CHIP:TOO:       VendorId: 65521
[1655710441.935784][27927:27928] CHIP:TOO:       FabricId: 1
[1655710441.935794][27927:27928] CHIP:TOO:       NodeId: 1
[1655710441.935804][27927:27928] CHIP:TOO:       Label:
[1655710441.935815][27927:27928] CHIP:TOO:       FabricIndex: 1
[1655710441.935825][27927:27928] CHIP:TOO:      }
...
[1655710441.937164][27927:27928] CHIP:TOO:     [15]: {
[1655710441.937175][27927:27928] CHIP:TOO:       RootPublicKey: 040535C04FF8D9105DBAFF9F940DE6D04CEA09F292A354A32E92636771C65568BE896F454E9B524FD1797C399AD388335574855806
[1655710441.937185][27927:27928] CHIP:TOO:       .............: AE1B434D3809C2F07937831E
[1655710441.937194][27927:27928] CHIP:TOO:       VendorId: 65521
[1655710441.937203][27927:27928] CHIP:TOO:       FabricId: 24
[1655710441.937212][27927:27928] CHIP:TOO:       NodeId: 1
[1655710441.937220][27927:27928] CHIP:TOO:       Label:
[1655710441.937230][27927:27928] CHIP:TOO:       FabricIndex: 15
[1655710441.937238][27927:27928] CHIP:TOO:      }

bzbarsky-apple, 2022-06-20:
@sstruchtrup Do you have a device that has both of the following:

1. `CHIP_CONFIG_MAX_FABRICS` set to 16
2. `CHIP_CONFIG_PEER_CONNECTION_POOL_SIZE` set to 16

?  The latter (the secure session pool size) has 16 as the default value on the pre-SVE commit you reference above....

If that's the setup, then it would fail as described because you'd need to have 16 CASE sessions and 1 PASE session to commission 16 fabrics, assuming no session eviction (which is not implemented yet).

On tip, the session pool size is called `CHIP_CONFIG_SECURE_SESSION_POOL_SIZE` and defaults to `(CHIP_CONFIG_MAX_FABRICS * 3 + 2)` so this issue should not be a problem there...

sstruchtrup, 2022-06-21:
@bzbarsky-apple Thanks for the quick feedback. Increasing `CHIP_CONFIG_PEER_CONNECTION_POOL_SIZE` to 17 will actually allow 16 fabrics to be commissioned.

closed 2022-06-21 by sstruchtrup