project-chip/connectedhomeip #19769: [Pre SVE] TC_G-2.2 Removing groups doesnt work when DUT in multiple groups

HubertHugo, 2022-06-20:
SHA: 1e3f12041013dd71a53d035dbf2f6d142fc06c4e
platform: nRF52840
all clusters app

When i have DUT in at least 2 groups there is no way to remove groups without at least one of them becoming irremovable.
First scenario:
1. Add 2 groups
2. Remove all groups with remove-all-groups command
Second scenario:
1. Add 2 groups
2. remove one of them with remove-group command

Further removing has no impact on groups. You can add endpoints to them but you can no longer get rid of them.

First results in only endpoints being deleted from groups, and groups become irremovable.
Second removes one of groups but second group becomes irremovable (following remove command removes only endpoint from it)

[Groups_arent_removed_chip-tool.log](https://github.com/project-chip/connectedhomeip/files/8941198/Groups_arent_removed_chip-tool.log)
[Groups_arent_removed_2nd_scenario_chip-tool.log](https://github.com/project-chip/connectedhomeip/files/8941194/Groups_arent_removed_2nd_scenario_chip-tool.log)

sankarselvam, 2022-06-20:
Looks like Duplicate of https://github.com/project-chip/connectedhomeip/issues/19610

bzbarsky-apple, 2022-06-20:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/19610