project-chip/connectedhomeip #19791: [pre-sve] TC-CC-2.1 ColorCapabilities attribute doesn't have the same values as the corresponding bits of the FeatureMap attribute.

mkleszcz-no, 2022-06-21:
all-cluster-app
nrf52840dk
revison: https://github.com/project-chip/connectedhomeip/commit/1e3f12041013dd71a53d035dbf2f6d142fc06c4e

TC-CC-2.1, step 19:
 #### Problem
 * Expected behavior:
    Bits 0-4 of the ColorCapabilities attribute SHALL have the same values as the corresponding bits of the FeatureMap attribute.

 * Actual behavior:
   FeatureMap: 31
   color capabilities : 0

 * Logs:
[TC_CC_2_1.log](https://github.com/project-chip/connectedhomeip/files/8946924/TC_CC_2_1.log)
[TC_CC_2_1_DUT.log](https://github.com/project-chip/connectedhomeip/files/8946925/TC_CC_2_1_DUT.log)

bzbarsky-apple, 2022-06-21:
@jmartinez-silabs

woody-apple, 2022-06-21:
Cert Blocker Review: @jmartinez-silabs, can you take a peek please?

jmartinez-silabs, 2022-06-21:
I will fix it right away. I do wonder why this attribute is needed, as based on the spec, it is the exact same thing as the feature map.

bzbarsky-apple, 2022-06-22:
@jmartinez-silabs I expect it's Zigbee legacy that got carted along "for compat"...  Making bridges' lives easier by making everyone else's lives harder.

closed 2022-06-22 by jmartinez-silabs