project-chip/connectedhomeip #19794: [SQA] Door-State is constant in doorlock attributes

Survensa, 2022-06-21:
### Problem :
Door Position Sensor is not able to fetch current door-state, it remains constant when door-lock is unlocked from lock-state or locked from unlock-state. Due to this DUT is not able to fetch any door-state-change event.

## Platform used : 

- > Tested on : 2ac3ee761232b466db70c196986e90081f2c4ec3
- > server : lock-app ( RPI-4, 8GB RAM)
- > client : chip-tool ( RPI-4, 8GB RAM )
- > network : Ethernet (IP)

## Steps to Reproduce :

1. Checked for current lock state : `./chip-tool doorlock read lock-state 1 1`

> [1655799523.462278][11408:11413] CHIP:TOO:   LockState: 1
2.  Checked for current door state : `./chip-tool doorlock read door-state 1 1`

> [1655799671.986956][11421:11426] CHIP:TOO:   DoorState: 0
3. Unlocked door from lock state :` ./chip-tool doorlock unlock-door 1 1 --timedInteractionTimeoutMs 1000`

> [1655800200.270350][11456:11461] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0101 Command=0x0000_0001 Status=0x0
4. Checked lock-state after unlocking :` ./chip-tool doorlock read door-state 1 1`

> [1655800295.826025][11480:11485] CHIP:TOO:   LockState: 2

5. Checked door-state now :  `./chip-tool doorlock read door-state 1 1`

> [1655800406.609853][11544:11549] CHIP:TOO:   DoorState: 0

6. Checked door-state-chnage event :  `./chip-tool doorlock read-event door-state-change 1 1`

> **DUT response :** 
> [1655800790.809914][10675:10675] CHIP:DMG: Building Reports for ReadHandler with LastReportGeneration = 0 DirtyGeneration = 0
> [1655800790.811021][10675:10675] CHIP:DMG: Fetched 0 events
> [1655800790.811052][10675:10675] CHIP:DMG: <RE> Sending report (payload has 7 bytes)...

### Full log :
[Door_state log.txt](https://github.com/project-chip/connectedhomeip/files/8947256/Door_state.log.txt)

bzbarsky-apple, 2022-06-21:
I don't see anything in the spec that prevents the "door is open but lock is in the locked position" situation.  So staying in the "door open" state when lock-door happens does not seem impossible...

@DanM-AssaAbloy @RyanLindemanCAE  @Morozov-5F

woody-apple, 2022-06-21:
Cert Blocker Review: Agree with @DanM-AssaAbloy above. Marking invalid test and closing.

closed 2022-06-21 by woody-apple