project-chip/connectedhomeip #19801: Group Communication - Unable to verify if the multicast command is working correctly

sethunk, 2022-06-21:
**Problem** - We are sending the multicast command to DUT for writing the node-label attribute value to "newnode" in Basic Information Cluster. We are trying to read the same attribute value in unicast mode to check if the multicast command actually worked but we are not able to see the value changed in the attribute.
Before sending the multicast command we are setting the appropriate ACL entry and Group settings command to initialize the test condition.

****Sequence of the Commands sent:**
1.Initial Commands**

1. ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{"cluster": null, "endpoint": 0, "deviceType": null}]}, {"fabricIndex": 1, "privilege": 3, "authMode": 3, "subjects": null, "targets": [{"cluster": null, "endpoint": 1, "deviceType": null}]}]' 1 0
2. ./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 42,"groupKeySecurityPolicy": 0, "epochKey0":"d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 2220000,"epochKey1":"d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 2220001,"epochKey2":"d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 2220002 }' 1 0
3. ./chip-tool groups add-group 0x0001 grp1 1 0
4. ./chip-tool groupkeymanagement write group-key-map '[{"groupId": 1, "groupKeySetID": 42, "fabricIndex": 1}]' 1 0
5. ./chip-tool groups view-group 0x0001 1 0
6. ./chip-tool groupkeymanagement key-set-read 42 1 0
7. ./chip-tool groupkeymanagement read group-table 1 0
8. ./chip-tool groupsettings add-group grp1 0x0001
9. ./chip-tool groupsettings add-keysets 0x42 0 0x000000000021dfe hex:d0d1d2d3d4d5d6d7d8d9dadbdcdddedf
10. ./chip-tool groupsettings bind-keyset 0x0001 0x42
11. ./chip-tool groupsettings show-groups

******************************************************************************************************************************

**2.Sending Multicast commands**

12. ./chip-tool basic write node-label newnode 0xffffffffffff0001 0

******************************************************************************************************************************
**3.Reading on-off attribute**

13. ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{"cluster": null, "endpoint": 0, "deviceType": null}]}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets": [{"cluster": null, "endpoint": 0, "deviceType": null}]}]' 1 0

14. ./chip-tool basic read node-label 1 0

******************************************************************************************************************************
**4. Test Environment**

**App used** - allclusters app
**Platform** - Chip-tool - RPI-4, 8GB RAM
                       DUT - RPI-4, 8GB RAM
**Network** - Ble-wifi
**Commit** - 2ac3ee761232b466db70c196986e90081f2c4ec3

**5. Logs**
[DUT_Logs_21.06.2022.txt](https://github.com/project-chip/connectedhomeip/files/8947801/DUT_Logs_21.06.2022.txt)
[TH_Logs_21.06.2022.txt](https://github.com/project-chip/connectedhomeip/files/8947803/TH_Logs_21.06.2022.txt)

bzbarsky-apple, 2022-06-21:
Duplicate of #19608

closed 2022-06-21 by bzbarsky-apple

mkardous-silabs, 2022-06-22:
The correct ACL command to allow a groud write attribute command is :
```
chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 4, "authMode": 3, "subjects": [1], "targets": null }]' <nodeId> 0
```

Also, you do not need to send a second acl write before actually sending the group command.