project-chip/connectedhomeip #19807: [SQA] Lock-Operation-Error & Door Lock Alarm events need to be implemented in Lock-app.

Survensa, 2022-06-21:
### Summary : 

While identifing verification steps of doorlock test plan by SQA found that the following events are not able to be triggered.

## Events which needs implementation :

1. > LockOperationError Event for DisabledUserDenied and Restricted
2. > Door-lock-alarm Event for  LockJammed
3. > Door-state-change for Lock and Unlock [ https://github.com/project-chip/connectedhomeip/issues/19794 ]

## Platform used :
Tested on : https://github.com/project-chip/connectedhomeip/commit/2ac3ee761232b466db70c196986e90081f2c4ec3

server : lock-app ( RPI-4, 8GB RAM)

client : chip-tool ( RPI-4, 8GB RAM )

network : Ethernet (IP)

### Reference : 
- SPEC : https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/DoorLock.adoc#5-events

Survensa, 2022-06-21:
@Morozov-5F  As discussed, I have raised this issue, kindly give your comments if any.

woody-apple, 2022-06-21:
Cert Blocker Review: @Morozov-5F can you please take a look at the mandatory events? Thanks

cjandhyala, 2022-06-28:
@Morozov-5F For the DIAG clusters we had developers implemented some signal handlers to generate the events, Could we follow a similar mechanism to generate these events?

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

PR #21242 (2022-07-29): Add ability to send required events for Door Lock app
#### Problem
* Fixes #19807 by adding ability to send more events for Linux Door Lock app.

#### Change overview
1. Added ability to send commands to the lock-app over the named pipe. See [README.md](https://github.com/project-chip/connectedhomeip/blob/68769c9e34fd84f70c7893ba7e251a9f7fd6e75f/examples/lock-app/linux/README.md) for more details.
1. Now Lock App for Linux is able to restrict access for users of Schedule Restricted User based on schedules.

#### Testing
* Send `SetDoorState` command over the named pipe (see [README.md](https:/ [... cut]
PR #21439 (2022-07-30): Add ability to send required events for Door Lock app
#### Problem
* Fixes #19807 by adding ability to send more events for Linux Door Lock app.

#### Change overview
1. Added ability to send commands to the lock-app over the named pipe. See [README.md](https://github.com/project-chip/connectedhomeip/blob/68769c9e34fd84f70c7893ba7e251a9f7fd6e75f/examples/lock-app/linux/README.md) for more details.
1. Now Lock App for Linux is able to restrict access for users of Schedule Restricted User based on schedules.

#### Testing
* Send `SetDoorState` command over the named pipe (see [README.md](https:/ [... cut]
closed 2022-07-29 by woody-apple