project-chip/connectedhomeip #19810: [Pre-SVE] TC-SU 2.8 when OTA Provider commissioned first, does not let second commissioner commission to the DUT

HubertHugo, 2022-06-21:
SHA: 1e3f12041013dd71a53d035dbf2f6d142fc06c4e
platform: Linux
OTA Provider

When I commission DUT and OTA Provider, second commissioner isn't commissioning to DUT. OTA Provider doesn't change his CM value back to 0 after being commissioned.

[ota_commissioned_but_cm_is_1_chip-tool.log](https://github.com/project-chip/connectedhomeip/files/8949622/ota_commissioned_but_cm_is_1_chip-tool.log)

bzbarsky-apple, 2022-06-21:
@HubertHugo In particular, it looks like commissioning discovery found the wrong thing for that second chip-tool pairing command, but since that command said "find me anything on the network at all that advertises commissionable", and there are multiple commissionable advertisements shown by `avahi-browse`, that's somewhat expected.

It's possible that some of those commissionable advertisements are stale, because on the SHA you're on minimal mdns does not withdraw stale advertisements and they have nonzero TTLs, so get cached by DNS resolvers.

I strongly recommend pairing by discriminator (note that the discriminators for the devices involved are different) instead of issuing a "pair with whatever you find out there" command and hoping.  That is, instead of `pairing onnetwork` use `pairing onnetwork-long` and provide the relevant discriminator.

bzbarsky-apple, 2022-12-20:
@HubertHugo Is this still a problem?  Are you compiling with minimal mdns or platform mdns for the OTA Provider?

Is this reproducible with chip-tool as the commissioners?  If so, the exact commands you are running for all the relevant apps would be pretty helpful in trying to reproduce this.

HubertHugo, 2023-01-02:
Now it works as it should.

closed 2023-01-02 by HubertHugo