project-chip/connectedhomeip #19839: AddTrustedRootCertificate Command send constraint_error when invalid_command is expected while sending malformed certificate

sowmyassp, 2022-06-22:
Problem:
When Arm fail command is sent  with malformed Root Certificate in a new fail safe timer period we expex to see INVALID _Command but we are seeing Constraint error
Spec Referrence-11.18.7.13
[image]

Sequence of Steps:
1.Commission DUT to TH using below commands : 
2.

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-06-22:
I read through the test plan, and it looks like this s trying to do step 40 from the test plan?  But the steps being carried out above look nothing like the parts of the test plan that come before step 40....

closed 2022-06-22 by bzbarsky-apple

bzbarsky-apple, 2022-06-22:
Though following the test plan exactly would run into trouble too.  Filed https://github.com/CHIP-Specifications/chip-test-plans/issues/1560