project-chip/connectedhomeip #19840: [Pre-SVE] Test TC-SC-5.2 Update of onoff status and nodelabel does not work.

HubertHugo, 2022-06-22:
SHA: 1e3f12041013dd71a53d035dbf2f6d142fc06c4e
platform: nRF52840
all clusters app

I cant update onoff status and nodelabel using group communication.

First I did test 5.1 with commands:
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{"cluster": null, "endpoint": 0, "deviceType": null}]}, {"fabricIndex": 1, "privilege": 3, "authMode": 3, "subjects": null, "targets": [{"cluster": null, "endpoint": 1, "deviceType": null}]}]' 1 0
./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 42,
"groupKeySecurityPolicy": 0, "epochKey0":
"d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 2220000,"epochKey1":
"d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 2220001,"epochKey2":
"d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 2220002 }' 1 0
./chip-tool groups add-group 0x0001 grp1 1 0
./chip-tool groupkeymanagement write group-key-map '[{"groupId": 1, "groupKeySetID": 42, "fabricIndex": 1}]' 1 0
./chip-tool groups view-group 0x0001 1 0
./chip-tool groupkeymanagement key-set-read 42 1 0
./chip-tool groupkeymanagement read group-key-map 1 0
./chip-tool groupkeymanagement read group-table 1 0
./chip-tool groupsettings add-group grp1 0x0001
./chip-tool groupsettings add-keysets 0x0042 0 0x000000000021dfe0 hex:d0d1d2d3d4d5d6d7d8d9dadbdcdddedf
./chip-tool groupsettings bind-keyset 0x0001 0x0042
./chip-tool groupsettings show-groups

And when I try to run test 5.2 
after command
./chip-tool basic write node-label newnode 0xffffffffffff0001 0
command:
./chip-tool basic read node-label 1 0
returns empty label
also in following steps command
./chip-tool onoff read on-off 1 1 
returns false, and its not changed by
./chip-tool onoff on 0xffffffffffff0001 1

same case (at least with label) occurs when I run 
./chip-tool tests TestGroupMessaging --nodeId 1 
instead

Side note: ACLs in first step are bad and don't give you permissions establish case session on endpoint 1.

All these commands are from excel with verification steps.

mkardous-silabs, 2022-06-22:
@HubertHugo See this comment for steps that work for groups. 
https://github.com/project-chip/connectedhomeip/issues/19608#issuecomment-1156931777

and replace the acl command by thise one 
https://github.com/project-chip/connectedhomeip/issues/19801#issuecomment-1163223895

bzbarsky-apple, 2022-06-22:
> ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{"cluster": null, "endpoint": 0, "deviceType": null}]}, {"fabricIndex": 1, [...]

Allows any group access to endpoint ***1***.

>  ./chip-tool groups add-group 0x0001 grp1 1 0

Adds endpoint ***0*** to group 0x0001.

> ./chip-tool basic write node-label newnode 0xffffffffffff0001 0

Will fail ACL checks because Basic Information is on endpoint 0 (which is in the group), but we only have ACLs for endpoint 1.

>  ./chip-tool onoff on 0xffffffffffff0001 1

Will do nothing because only endpoint 0 is in the group and there is no On/Off cluster on that endpoint.

> All these commands are from excel with verification steps.

Looks like those steps need fixing.  I filed https://github.com/CHIP-Specifications/chip-test-plans/issues/1561

closed 2022-06-22 by bzbarsky-apple

mkardous-silabs, 2022-06-22:
@bzbarsky-apple  This is a duplicate of 
#19801  and #19608

bzbarsky-apple, 2022-06-22:
Yep, but finally someone explicitly said that it's the verification spreadsheet that's confused.... :)