project-chip/connectedhomeip #19844: TC-DIAG-TH_NW 2.1 Neighbor table list contains an entry with FrameErrorRate out of range, Security policy Flags value is also out of range

HubertHugo, 2022-06-22:
SHA: 1e3f12041013dd71a53d035dbf2f6d142fc06c4e
platform: nRF52840
all clusters app

Frame error rate should be in range 0-100 but it's 124.

./chip-tool threadnetworkdiagnostics read neighbor-table-list 1 0
```
[1655894245.353869][13550:13555] CHIP:TOO:   NeighborTableList: 1 entries
[1655894245.353936][13550:13555] CHIP:TOO:     [1]: {
[1655894245.353959][13550:13555] CHIP:TOO:       ExtAddress: 15490139605606181840
[1655894245.353971][13550:13555] CHIP:TOO:       Age: 19
[1655894245.353983][13550:13555] CHIP:TOO:       Rloc16: 8192
[1655894245.353993][13550:13555] CHIP:TOO:       LinkFrameCounter: 1947
[1655894245.354003][13550:13555] CHIP:TOO:       MleFrameCounter: 636
[1655894245.354023][13550:13555] CHIP:TOO:       Lqi: 3
[1655894245.354036][13550:13555] CHIP:TOO:       AverageRssi: -38
[1655894245.354047][13550:13555] CHIP:TOO:       LastRssi: -39
[1655894245.354058][13550:13555] CHIP:TOO:       FrameErrorRate: 124
[1655894245.354070][13550:13555] CHIP:TOO:       MessageErrorRate: 0
[1655894245.354081][13550:13555] CHIP:TOO:       RxOnWhenIdle: TRUE
[1655894245.354091][13550:13555] CHIP:TOO:       FullThreadDevice: TRUE
[1655894245.354101][13550:13555] CHIP:TOO:       FullNetworkData: TRUE
[1655894245.354107][13550:13555] CHIP:TOO:       IsChild: FALSE
[1655894245.354117][13550:13555] CHIP:TOO:      }
```

Flags should be of type uint8 but its 8335.

./chip-tool threadnetworkdiagnostics read security-policy 1 0
```
[1655895190.363600][14037:14042] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0035 Attribute 0x0000_003B DataVersion: 1367807082
[1655895190.363619][14037:14042] CHIP:TOO:   SecurityPolicy: {
[1655895190.363623][14037:14042] CHIP:TOO:     RotationTime: 672
[1655895190.363626][14037:14042] CHIP:TOO:     Flags: 8335
[1655895190.363629][14037:14042] CHIP:TOO:    }
```

bzbarsky-apple, 2022-06-22:
For the neighbor table frame error rate `src/platform/OpenThread/GenericThreadStackManagerImpl_OpenThread.cpp` does:
```
neighborTable.frameErrorRate   = neighInfo.mFrameErrorRate;
```
where `neighInfo` is a `otNeighborInfo`.  Its definition for `mFrameErrorRate` at https://openthread.io/reference/struct/ot-neighbor-info#structot_neighbor_info_1a98814b49467b869ff4a15bba5c1d9ca2 says:
> uint16_t otNeighborInfo::mFrameErrorRate
> Frame error rate (0xffff->100%). 

I'm surprised this assignment even compiles.  @Damian-Nordic @kkasperczyk-no is the relevant directory not using `-Wconversion` in its build flags?

For the security policy, it looks like our XML is wrong:
```
  <struct name="SecurityPolicy">
    <cluster code="0x0035"/>
    <item name="RotationTime" type="INT16U"/>
    <item name="Flags" type="BITMAP16"/>
  </struct>
```
(as noted above per spec the flags are `uint8`) and the implementation in `GenericThreadStackManagerImpl_OpenThread<ImplClass>::_WriteThreadNetworkDiagnosticAttributeToTlv` does not make sense to me:
```
            memcpy(&policyAsInts, &activeDataset.mSecurityPolicy, sizeof(policyAsInts));
...
            securityPolicy.flags        = policyAsInts[1];
```
but `mSecurityPolicy` is defined at https://openthread.io/reference/struct/ot-security-policy to have 9 boolean members, a uint8_t member, and a uint16_t member (rotation time), and nothing I see says that they come in the order this code is expecting....  And clearly we are getting bits that are not part of the actual flags in some way here.

kkasperczyk-no, 2022-06-27:
The spec issue: https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5414

closed 2022-06-27 by andy31415