project-chip/connectedhomeip #19851: [Pre-SVE] User label writing too long label

HubertHugo, 2022-06-22:
SHA: 1e3f12041013dd71a53d035dbf2f6d142fc06c4e
platform: nRF52840
all clusters app

When doing test TC-ULABEL-2.2 i wrote this command to see what happens:
 ./chip-tool userlabel write label-list '[{"label":"roomName", "value":"master bedroom 1"}, {"label":"orientationorientation", "value":"east"}, {"label":"floor", "value":"2"}, {"label":"roomType", "value":"bedroom"}]' 1 0
 
So one of labels is longer than limit (16 bytes)
I got SUCCESS and when used
./chip-tool userlabel read label-list 1 0
I got:
```
[1655903432.210268][17776:17781] CHIP:TOO:   label list: 4 entries
[1655903432.210275][17776:17781] CHIP:TOO:     [1]: {
[1655903432.210278][17776:17781] CHIP:TOO:       Label: roomName
[1655903432.210281][17776:17781] CHIP:TOO:       Value: master bedroom 1
[1655903432.210284][17776:17781] CHIP:TOO:      }
[1655903432.210288][17776:17781] CHIP:TOO:     [2]: {
[1655903432.210291][17776:17781] CHIP:TOO:       Label: orientationorien
[1655903432.210295][17776:17781] CHIP:TOO:       Value: east
[1655903432.210297][17776:17781] CHIP:TOO:      }
[1655903432.210302][17776:17781] CHIP:TOO:     [3]: {
[1655903432.210304][17776:17781] CHIP:TOO:       Label: floor
[1655903432.210306][17776:17781] CHIP:TOO:       Value: 2
[1655903432.210309][17776:17781] CHIP:TOO:      }
[1655903432.210312][17776:17781] CHIP:TOO:     [4]: {
[1655903432.210315][17776:17781] CHIP:TOO:       Label: roomType
[1655903432.210317][17776:17781] CHIP:TOO:       Value: bedroom
[1655903432.210319][17776:17781] CHIP:TOO:      }
```
So the label was just cropped. Shouldn't I get some kind of error while writing instead?

HubertHugo, 2022-06-22:
Later I found Test TC-ULABEL 2.3 tries to test it but does not handle it properly. From what I found it let's you assign Label and Value that are combined not longer than 34 characters. No matter the length read displays first 16 characters from both.

bzbarsky-apple, 2022-06-22:
In `examples/providers/DeviceInfoProviderImpl.cpp` we have `DeviceInfoProviderImpl::SetUserLabelAt` which will happily store over-size values as long as they fit in the total space it has for the TLV.  But on read we go through `DeviceInfoProviderImpl::UserLabelIteratorImpl::Next` which does:
```
    Platform::CopyString(mUserLabelNameBuf, label);
    Platform::CopyString(mUserLabelValueBuf, value);
```
which will silently truncate lengths.

This needs to be fixed to be spec-compliant and reject over-long things on write.

bzbarsky-apple, 2022-06-22:
@yufengwangca @andy31415

cjandhyala, 2022-06-28:
Per Rajashree, this TC TC-ULABEL 2.3 passes on raspi , all-cluster-app. Looks like issue specific to nordic platform

bzbarsky-apple, 2022-06-28:
I don't think it is.  I reproduced the problem on the "linux" version of all-clusters-app locally...  @cjandhyala can you please double-check with Rajashree _exactly_ what was tested?

Rajashreekalmane, 2022-06-30:
Sorry @cjandhyala @bzbarsky-apple my bad, As per test plan while verifying i given less than 16 byte for label attribute ...If try more than 16 to till 30 getting success response instead failure, but <30 if  I try to write getting failure response.

woody-apple, 2022-07-06:
SDK Review: @bzbarsky-apple are you able to take a look at this when you're back? Thanks!

closed 2022-07-08 by woody-apple

andy31415, 2022-07-11:
Note #20552 - we now fail the write, however the error code is not fully correct.

PR #20431 (2022-07-08): Enforce constraints for sizes of labelstruct label and value lengths
#### Problem

Server will allow structs with members larger than 16 characters, but reading the values back will get truncated.

Fixes #19851

#### Change overview
Add constraint checking into our writer logic.

#### Testing
Manually tried to write an overly large value, saw back a FAILURE response.
Added a yaml integration test
PR #20501 (2022-07-11): Enforce constraints for sizes of labelstruct label and value lengths
#### Problem

Server will allow structs with members larger than 16 characters, but reading the values back will get truncated.

Fixes #19851

#### Change overview
Add constraint checking into our writer logic.

#### Testing
Manually tried to write an overly large value, saw back a FAILURE response.
Added a yaml integration test