project-chip/connectedhomeip #19869: Test_TC_LVL_6_1.yaml has disabled test step

woody-apple, 2022-06-23:
#### Problem

This YAML step is failing per example here: https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true

```
2022-06-23 03:24:50.401 ERROR   03:24:47.707 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [EM] Received message of type 0x5 with protocolId (0, 1) and MessageCounter:171556752 on exchange 32088i
[2662](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2663)
2022-06-23 03:24:50.401 ERROR   03:24:47.707 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [EM] Found matching exchange: 32088i, Delegate: 0x7b3c000177f0
[2663](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2664)
2022-06-23 03:24:50.401 ERROR   03:24:47.707 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [EM] Rxd Ack; Removing MessageCounter:213441215 from Retrans Table on exchange 32088i
[2664](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2665)
2022-06-23 03:24:50.401 ERROR   03:24:47.707 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [EM] Removed CHIP MessageCounter:213441215 from RetransTable on exchange 32088i
[2665](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2666)
2022-06-23 03:24:50.401 ERROR   03:24:47.707 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [DMG] ReportDataMessage =
[2666](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2667)
2022-06-23 03:24:50.401 ERROR   03:24:47.707 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [DMG] {
[2667](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2668)
2022-06-23 03:24:50.401 ERROR   03:24:47.707 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [DMG] 	AttributeReportIBs =
[2668](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2669)
2022-06-23 03:24:50.401 ERROR   03:24:47.707 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [DMG] 	[
[2669](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2670)
2022-06-23 03:24:50.401 ERROR   03:24:47.707 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [DMG] 		AttributeReportIB =
[2670](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2671)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [DMG] 		{
[2671](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2672)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [DMG] 			AttributeDataIB =
[2672](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2673)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [DMG] 			{
[2673](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2674)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687707] [26076:104009] CHIP: [DMG] 				DataVersion = 0x61a4d190,
[2674](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2675)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 				AttributePathIB =
[2675](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2676)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 				{
[2676](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2677)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 					Endpoint = 0x1,
[2677](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2678)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 					Cluster = 0x8,
[2678](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2679)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 					Attribute = 0x0000_0000,
[2679](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2680)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 				}
[2680](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2681)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 					
[2681](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2682)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 				Data = 26, 
[2682](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2683)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 			},
[2683](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2684)
2022-06-23 03:24:50.402 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 			
[2684](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2685)
2022-06-23 03:24:50.403 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 		},
[2685](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2686)
2022-06-23 03:24:50.403 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 		
[2686](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2687)
2022-06-23 03:24:50.403 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 	],
[2687](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2688)
2022-06-23 03:24:50.403 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 	
[2688](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2689)
2022-06-23 03:24:50.403 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 	SuppressResponse = true, 
[2689](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2690)
2022-06-23 03:24:50.403 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] 	InteractionModelRevision = 1
[2690](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2691)
2022-06-23 03:24:50.403 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [DMG] }
[2691](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2692)
2022-06-23 03:24:50.403 ERROR   03:24:47.708 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [TOO]  ***** Test Failure: currentLevel value mismatch: expected 25 but got 26
[2692](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2693)
2022-06-23 03:24:50.403 ERROR   03:24:47.708 - TEST OUT  : 
[2693](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2694)
2022-06-23 03:24:50.403 ERROR   03:24:47.709 - TEST OUT  : [1655954687708] [26076:104009] CHIP: [EM] Sending Standalone Ack for MessageCounter:171556752 on exchange 32088i
[2694](https://github.com/project-chip/connectedhomeip/runs/7015795479?check_suite_focus=true#step:11:2695)
```

The YAML is:
```yaml
    - label: "Reads CurrentLevel attribute from DUT"
      disabled: true
      command: "readAttribute"
      attribute: "current level"
      response:
          constraints:
              notValue: CurrentLevelValue
```

 #### Proposed Solution
Fix

kvikrambhat, 2022-06-23:
@raju-apple @woody-apple The test steps that is in question here is marked by the PICS "[PICS_SKIP_SAMPLE_APP](https://github.com/project-chip/connectedhomeip/blob/29451fb276acd27d911ab7023d124a4965cfe04d/src/app/tests/suites/certification/PICS.yaml#L615)" , so this steps should not even run on the CI as this PIC was meant to skip steps that are currently not supported by the sample apps.

manjunath-grl, 2022-06-28:
We are observing inconsistent values for the CurrentLevel attribute after the wait time, instead of checking for specific value we have updated the script to check for a range with 5% tolerance

PR #20030 (2022-06-28): Use 5% tolerance when checking CurrentLevel attribute in TC-LVL-6.1
Modified script TC-LVl-6.1
We are observing inconsistent values for the CurrentLevel attribute after the wait time, instead of checking for specific value we have updated the script to check for a range with 5% tolerance
Added auto generated files

Attached execution logs for reference:
[Execution_log.txt](https://github.com/project-chip/connectedhomeip/files/9000870/Execution_log.txt)

Fixes #19869
closed 2022-06-28 by woody-apple