project-chip/connectedhomeip #19872: CHIPDeviceControllerFactory.cpp:245:5: unimplemented: non-trivial designated initializers not supported

tatsutaigu, 2022-06-23:
Hi, 
I tried below example.
[https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32](https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32)

I am using Jetson Nano as Linux PC for developing.
However, I cannot build chip-tool.
How do I resolve it?

> $ git log -n 10
> commit 0a995b0a1fd1f76efea9308a16792ae4f318781f (HEAD -> master, origin/master, origin/HEAD)
> Author: Daniel Kneip <80306648+latch-danielkneip@users.noreply.github.com>
> Date:   Wed Jun 22 19:12:24 2022 -0700
> 
>     Linux Matter bridge app that supports Wi-Fi without Thread (#19725)
> 
>     * [Latch] For Matter SDK bridge, support Wi-Fi without needing Thread (#18782)
> 
>     * Restyled by gn
> 
>     * [Latch] For Matter SDK bridge, support Wi-Fi without needing Thread (#18782)
> 
>     * Restyled by clang-format
> 
>     Co-authored-by: Restyled.io <commits@restyled.io>

> $ cd ~/connectedhomeip/
> 
> $ source ./scripts/activate.sh
> Activating environment (setting environment variables):
> 
>   Setting environment variables for CIPD package manager...done
>   Setting environment variables for Python environment.....done
>   Setting environment variables for pw packages............skipped
>   Setting environment variables for Host tools.............done
> 
> Checking the environment:
> 
> 20220623 12:54:16 INF Generating grammar tables from /usr/lib/python3.8/lib2to3/Grammar.txt
> 20220623 12:54:16 INF Generating grammar tables from /usr/lib/python3.8/lib2to3/PatternGrammar.txt
> 20220623 12:54:18 INF Environment passes all checks!
> 
> Environment looks good, you are ready to go!
> 
> $ cd examples/chip-tool
> 
> $ rm -rf out
> 
> $ gn gen out/debug
> Done. Made 104 targets from 108 files in 436ms
> 
> $ ninja -C out/debug
> ninja: Entering directory `out/debug'
> [252/501] c++ obj/third_party/connectedhomeip/src/controller/libChipController.CHIPDeviceControllerFactory.cpp.o
> FAILED: obj/third_party/connectedhomeip/src/controller/libChipController.CHIPDeviceControllerFactory.cpp.o
> aarch64-linux-gnu-g++ -MMD -MF obj/third_party/connectedhomeip/src/controller/libChipController.CHIPDeviceControllerFactory.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-unknown-warning-option -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wno-cast-function-type -fdiagnostics-color -fno-strict-aliasing -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -DCHIP_MINMDNS_HIGH_VERBOSITY=1 -I../../third_party/connectedhomeip/zzz_generated -I../../third_party/connectedhomeip/src/include -I../../third_party/connectedhomeip/src -Igen/include -I../../third_party/connectedhomeip/zzz_generated/app-common -I../../third_party/connectedhomeip/examples/chip-tool/include -I../../third_party/connectedhomeip/config/standalone -I../../third_party/connectedhomeip/third_party/nlassert/repo/include -I../../third_party/connectedhomeip/third_party/nlio/repo/include -I../../third_party/connectedhomeip/third_party/inipp/repo/inipp -c ../../third_party/connectedhomeip/src/controller/CHIPDeviceControllerFactory.cpp -o obj/third_party/connectedhomeip/src/controller/libChipController.CHIPDeviceControllerFactory.cpp.o
> ../../third_party/connectedhomeip/src/controller/CHIPDeviceControllerFactory.cpp: In member function ‘CHIP_ERROR chip::Controller::DeviceControllerFactory::InitSystemState(chip::Controller::FactoryInitParams)’:
> ../../third_party/connectedhomeip/src/controller/CHIPDeviceControllerFactory.cpp:245:5: sorry, unimplemented: non-trivial designated initializers not supported
>      };
>      ^
> ../../third_party/connectedhomeip/src/controller/CHIPDeviceControllerFactory.cpp:245:5: sorry, unimplemented: non-trivial designated initializers not supported
> ../../third_party/connectedhomeip/src/controller/CHIPDeviceControllerFactory.cpp:245:5: sorry, unimplemented: non-trivial designated initializers not supported
> ../../third_party/connectedhomeip/src/controller/CHIPDeviceControllerFactory.cpp:245:5: sorry, unimplemented: non-trivial designated initializers not supported
> ../../third_party/connectedhomeip/src/controller/CHIPDeviceControllerFactory.cpp:245:5: sorry, unimplemented: non-trivial designated initializers not supported
> [257/501] c++ obj/chip-tool.main.cpp.o
> ninja: build stopped: subcommand failed.
> 
> $

bzbarsky-apple, 2022-06-23:
@tatsutaigu What compiler ends up being used in your case?

I _suspect_ that switching the order of initializers to match the order of member declarations might work.  As does using a newer compiler...

tatsutaigu, 2022-06-23:
@bzbarsky-apple 
Thank you for your reply.
My environment only supports up to gcc-8/g++-8.
I guess it is not recommended version for this project.
I will change Jetson Nano to Raspberry Pi.

closed 2022-06-23 by tatsutaigu