project-chip/connectedhomeip #19898: ACL events not generated by AddNOC/RemoveFabric

tcarmelveilleux, 2022-06-23:
#### Problem
ACL events are not generated by `AddNOC` and `RemoveFabric` command handlers, which both cause the ACL to be mutated.

 #### Proposed Solution
- Find a way to centralize event notification

woody-apple, 2022-07-20:
Spec Review: @andy31415 to follow up, appears to be v1.0 if this is still an issue.

yufengwangca, 2022-08-02:
From the log, I see when the  AddNOC command is handled,  ACL an AccessControlEntryChangeddo event get logged.

```
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ ./chip-all-clusters-app | grep aaaa
[1659461311.391696][161082:161082] CHIP:ZCL: aaaa: OpCreds: Received an AddNOC command
[1659461311.394670][161082:161082] CHIP:DMG: aaaa: OnEntryChanged
[1659461311.394682][161082:161082] CHIP:DMG: aaaa:OnEntryChanged:kAdded
[1659461311.394690][161082:161082] CHIP:DMG: aaaa:OnEntryChanged:AuthMode::kPase
[1659461311.394747][161082:161082] CHIP:DMG: aaaa:OnEntryChanged:LogEvent:1

```
I see when the  RemoveFabric ommand is handled,  ACL an AccessControlEntryChangeddo event is not logged because subjectDescriptor is null in this case

```
[1659461323.195252][161082:161082] CHIP:ZCL: aaaa: OpCreds: Received a RemoveFabric Command for FabricIndex 0x1
[1659461323.201819][161082:161082] CHIP:DMG: aaaa:AccessControlCluster: OnEntryChanged
[1659461323.201831][161082:161082] CHIP:DMG: aaaa:OnEntryChanged:subjectDescriptor == nullptr
```
That means the entry was changed internally by the system (e.g. deleting entries when a/ fabric is removed), then there won't be a subject descriptor, and also it won't be appropriate to create an event. I think this is expected behavior, no?

tcarmelveilleux, 2022-08-02:
Not having ACL events on RemoveFabric is the expected behavior in my read, since the fabric removal would lead to a Leave event (which any client subscribed could get) and the fact that events on ACL are fabric-sensitive would mean they are not going to be seen by anything/anyone else.

bzbarsky-apple, 2022-08-02:
> and the fact that events on ACL are fabric-sensitive would mean they are not going to be seen by anything/anyone else.

Yes, agreed.

tcarmelveilleux, 2022-08-03:
Need to remove the TODOs

PR #21591 (2022-08-03): Remove TODO item which is no more valid
#### Problem
What is being fixed?  Examples:
* From the log, I see when the AddNOC command is handled, ACL an AccessControlEntryChangeddo event get logged.
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ ./chip-all-clusters-app | grep aaaa
[1659461311.391696][161082:161082] CHIP:ZCL: aaaa: OpCreds: Received an AddNOC command
[1659461311.394670][161082:161082] CHIP:DMG: aaaa: OnEntryChanged
[1659461311.394682][161082:161082] CHIP:DMG: aaaa:OnEntryChanged:kAdded
[1659461311.394690][161082:161082] CHIP:DMG: aaaa:OnEntryChanged:AuthMode::kPase
 [... cut]
closed 2022-08-03 by woody-apple