project-chip/connectedhomeip #19912: [pre-sve] [TC-DRLK-2.6] set-holiday-schedule results in failure

mamayorg, 2022-06-23:
**Problem**
When trying to set a holiday schedule, the return response is FAILURE.

**Command**
Commands used
./chip-tool doorlock read number-of-holiday-schedules-supported 1 1
./chip-tool doorlock set-holiday-schedule 1 20 30 0 1 1
./chip-tool doorlock get-holiday-schedule 1 1 1

 * expected behavior : Success response. Return the holiday schedule with the localStarttime and localEndTime as well as operating mode.
 * actual behavior: command gives FAILURE. the get command has missing information 
 
Commit
commit id: 1e3f12041013dd71a53d035dbf2f6d142fc06c4e
Platform: EFR32

[TC_DRLK_2.6_.txt](https://github.com/project-chip/connectedhomeip/files/8969529/TC_DRLK_2.6_.txt)

bzbarsky-apple, 2022-06-23:
>  Platform: EFR32

@mamayorg What is the server app being used?

bzbarsky-apple, 2022-06-23:
@Morozov-5F

mamayorg, 2022-06-28:
> > Platform: EFR32
> 
> @mamayorg What is the server app being used?

I'm using lock-app

Survensa, 2022-06-29:
@bzbarsky-apple I dont see this issue with RPI 

- Commit used : 81c7f2aacebbe5bb304a62ecdbeaff9220d72c69
- Chip-tool : RPI 
- DUT : Door lock app
- Network : IP (ethernet)

Full execution log of TC-DRLK-2.6 
[TC-DRLK-2.6 LOG](https://github.com/project-chip/connectedhomeip/files/9010043/TC-DRLK-2.6.LOG)

woody-apple, 2022-06-29:
Cert Blocker Review: @Morozov-5F , can you also take a look at this lock issue? Thanks!

bzbarsky-apple, 2022-06-29:
@Survensa which "Door lock app" are you using for DUT?  That is, which platform?

Survensa, 2022-06-30:
> @Survensa which "Door lock app" are you using for DUT?  That is, which platform?

Used RPI for both Chip-tool and DUT with lock app.

bzbarsky-apple, 2022-06-30:
OK, so that's the Linux lock app.  The original issue is reported with the efr32 lock app, which may have quite different behavior here; a lot of the logic in the lock apps is not shared.

bzbarsky-apple, 2022-06-30:
@mykrupp If this is EFR32-specific, could you take a look, please?

mykrupp, 2022-06-30:
The commit hash tested initially is likely missing parts from EFR32 side and door-lock-server side. I believe the door-lock-server holiday is working now.

This open PR contains schedules implementation on the EFR32 side.  I am hoping to get this PR merged before an SVE hash is cut today: https://github.com/project-chip/connectedhomeip/pull/20044

Morozov-5F, 2022-07-13:
Seems like this was fixed in the #20044 as it was EFR32-specific. Holiday schedules seem to be working in the lock-app under Linux, can't confirm for EFR32 as I lack the required hardware.

woody-apple, 2022-07-13:
Cert Blocker Review: @mamayorg can you please re-verify this, as we believe this is resolved.

closed 2022-07-13 by woody-apple