project-chip/connectedhomeip #19919: [pre-sve][TC-DRLK-2.5][TC-DRLK-2.7] Set-week-day-schedule and clear-week-day-schedule command returns NOT_FOUND

mamayorg, 2022-06-23:
**Problem**
when using set-week-day-schedule or clear-week-day-schedule commands, the return response is NOT_FOUND.

 **Commands**
Commands used:
./chip-tool doorlock read number-of-week-day-schedules-supported-per-user 1 1
./chip-tool doorlock read number-of-total-users-supported 1 1
./chip-tool doorlock set-week-day-schedule 1 1 2  15 45 16 55 1 1      (NOT_FOUND)
./chip-tool doorlock get-week-day-schedule 1 1 1 1
./chip-tool doorlock set-week-day-schedule 0 1 7 15 45 16 55 1 1
./chip-tool doorlock get-week-day-schedule 0 1 1 1
./chip-tool doorlock clear-week-day-schedule 1 1 1 1                           (NOT_FOUND)

 * expected behavior: response SUCCESS.
 * actual behavior: returns response NOT_FOUND. get-week-day-schedule missing startHour, endHour, startMinute, endMinute as well as dayMask

**Commit**
commit id: 1e3f12041013dd71a53d035dbf2f6d142fc06c4e
Platform: EFR32
Network: Thread

[TC_DRLK_2.5_.txt](https://github.com/project-chip/connectedhomeip/files/8970799/TC_DRLK_2.5_.txt)
[TC_DRLK_2.5_step#7.txt](https://github.com/project-chip/connectedhomeip/files/8970836/TC_DRLK_2.5_step.7.txt)
[TC_DRLK_2.5_step#3.txt](https://github.com/project-chip/connectedhomeip/files/8970838/TC_DRLK_2.5_step.3.txt)

bzbarsky-apple, 2022-06-23:
@mamayorg How exactly is this different from #19622, which is referenced above and seems like exactly the same issue?

Survensa, 2022-06-29:
@mamayorg & @bzbarsky-apple 

I am not reproducing this issue.

- Commit used : 81c7f2aacebbe5bb304a62ecdbeaff9220d72c69
- Chip-tool : RPI 
- DUT : Door lock app
- Network : IP (ethernet)

### Steps I followed for TC-DRLK-2.5 :

Step 1 :  As per pre-condition created new user 
`./chip-tool doorlock set-user 0 1 xxx 6452 1 0 0 1 1  --timedInteractionTimeoutMs 1000`

> [1656501301.674438][26350:26355] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0101 Command=0x0000_001A Status=0x0

Step 2 :  `./chip-tool doorlock read number-of-week-day-schedules-supported-per-user 1 1`

> [1656501436.394716][26366:26371] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0101 Attribute 0x0000_0014 DataVersion: 3514039033
> [1656501436.394837][26366:26371] CHIP:TOO:   NumberOfWeekDaySchedulesSupportedPerUser: 10

Step 3 : `./chip-tool doorlock read number-of-total-users-supported 1 1`

> [1656501476.501232][26373:26378] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0101 Attribute 0x0000_0011 DataVersion: 3514039033
> [1656501476.501461][26373:26378] CHIP:TOO:   NumberOfTotalUsersSupported: 10

Step 4 :` ./chip-tool doorlock set-week-day-schedule 1 1 2  15 45 16 55 1 1`

> [1656501520.689931][26381:26386] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0101 Command=0x0000_000B Status=0x0

Step 5 :` ./chip-tool doorlock get-week-day-schedule 1 1 1 1`

> [1656501601.516982][26390:26395] CHIP:DMG: Received Command Response Data, Endpoint=1 Cluster=0x0000_0101 Command=0x0000_000C
> [1656501601.517034][26390:26395] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0101 Command 0x0000_000C
> [1656501601.517108][26390:26395] CHIP:TOO:   GetWeekDayScheduleResponse: {
> [1656501601.517149][26390:26395] CHIP:TOO:     weekDayIndex: 1
> [1656501601.517173][26390:26395] CHIP:TOO:     userIndex: 1
> [1656501601.517197][26390:26395] CHIP:TOO:     status: 0
> [1656501601.517234][26390:26395] CHIP:TOO:     daysMask: 2
> [1656501601.517260][26390:26395] CHIP:TOO:     startHour: 15
> [1656501601.517360][26390:26395] CHIP:TOO:     startMinute: 45
> [1656501601.517426][26390:26395] CHIP:TOO:     endHour: 16
> [1656501601.517450][26390:26395] CHIP:TOO:     endMinute: 55
> [1656501601.517472][26390:26395] CHIP:TOO:    }

woody-apple, 2022-06-29:
Cert Blocker Review: @Morozov-5F Are you able to take look at this lock issue? Can you confirm this is a distinct issue?

bzbarsky-apple, 2022-06-29:
> Step 1 : As per pre-condition created new user

Yes, that's the difference: the steps listed in this issue at start do _not_ do that.  And if you don't do that, then the SDK behavior does not follow the spec.

bzbarsky-apple, 2022-06-29:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/19622

closed 2022-06-29 by bzbarsky-apple