project-chip/connectedhomeip #19923: Generic Switch: Linux sample uses previousPosition / newPosition wrongly and uses positions > numberOfPositions

markus-becker-tridonic-com, 2022-06-23:
#### Problem

previousPosition / newPosition handled wrongly in https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/linux/main-common.cpp#L159-L160.

Values of 10 and 20 are also wrong (should be 0 and 1), since numberOfPositions is 2: https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/all-clusters-common/all-clusters-app.matter#L4228.

#### Proposed Solution

Use values 0 and 1. Use newPosition=1 on press and previousPosition=1 on release.

woody-apple, 2022-07-20:
Spec Review: Reference app should be fixed for 1.0

markus-becker-tridonic-com, 2022-07-27:
previousPosition and newPosition are passed as parameters in JSON now and are no longer hard-coded anymore.

closed 2022-07-27 by markus-becker-tridonic-com