project-chip/connectedhomeip #19936: Make sure Leave event is emitted on internal deletions of fabrics

tcarmelveilleux, 2022-06-24:
#### Problem
The Basic Info cluster Leave event is required to be generated when a fabric is removed. It is not certain this is generated in every instance of fabric removal.

 #### Proposed Solution
- Audit all paths of fabric removal to ensure Leave event is generated.

bzbarsky-apple, 2022-07-25:
https://github.com/project-chip/connectedhomeip/issues/21012 tracks the ordering issue there.

bzbarsky-apple, 2023-04-25:
We now generate `Leave` events in `FabricWillBeRemoved`, which is a notification from the fabric table.  So any fabric removal that the fabric table knows about will generate the event.

@tcarmelveilleux Are you worried about fabric removals directly from storage, bypassing the fabric table?